pybluez/pybluez #232: Cannot install on Mac OS 10.14.2

anhmiuhv, 2019-02-05:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**Mac OS X 10.14.2
 - **Hardware:**Macbook Pro
 - **Python Version:**3.7
 - **PyBluez Version:**I build using the github link
pip install --upgrade --force-reinstall git+https://github.com/pybluez/pybluez.git#egg=pybluez

## Issue

Fail to install.
Log file: https://gist.github.com/anhmiuhv/40de2c97fc56de145a0d6216dbc1424a

rmoriz, 2019-03-08:
PR #223 fixes that issue for me

```shell
git clone https://github.com/pybluez/pybluez 
cd pybluez
curl https://patch-diff.githubusercontent.com/raw/pybluez/pybluez/pull/223.patch | patch -p1
python3 setup.py install 
```

uraimo, 2019-04-17:
Can we merge #223?

robertoostenveld, 2019-04-20:
The previous comment re #223 also solved it for me on macOS 10.13.6 and Python 2.7

rgov, 2019-04-20:
Merged #223.

closed 2019-04-20 by rgov