pybluez/pybluez #233: Distribute wheels for Linux?

Peque, 2019-02-25:
Not sure if it is possible or which are the challenges involved, but it would be great to have binary wheels distributed for PyBluez in Linux.

Peque, 2019-07-23:
Friendly ping. :innocent:

rgov, 2019-07-23:
This is something we should do if we're going to do it for other platforms.

beadysea, 2019-07-23:
Is there a plan to also have binary wheels for windows?

rgov, 2019-07-23:
Yes that's #180.

beadysea, 2019-07-24:
Sign posting this article which may provide some help in automating wheel building.
https://tomassetti.me/continous-integration-on-linux-and-windows-travis-and-appveyor/

rgov, 2019-12-28:
Currently PyPI does not allow us to submit a Linux wheel, but we may be able to follow PEP 513. A patch would be welcome.

laksh225, 2020-05-22:
I think wheels for Linux are supported now.