pybluez/pybluez #234: Pyblez doesn't work on Ubuntu

TziEm26, 2019-03-16:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - Ubuntu 16.04 LTS:
 -  Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
 - Python 2.7.12
 - Pybluez 0.18

## Issue

I have a Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode) plugged in a usb port of a Lenovo Ideapad 100 (laptop doesn't provide integrated Bluetooth service) . I dual boot Windows 7 and Ubuntu 16.04 LTS. I want to get data from a bluetooth device via a python script. When I use Windows 7 everything's working fine. I'm running a simple scan script:

Python 2.7.12

# simple inquiry example
import bluetooth
nearby_devices = bluetooth.discover_devices(lookup_names=True)
print("found %d devices" % len(nearby_devices))

for addr, name in nearby_devices:
    print("  %s - %s" % (addr, name)

However, when I turn to Ubuntu, despite the fact that bluetoothctl works fine, pybluez cant find any devices (found 0 devices). Just in case, I run on terminal pip install pybluez and got this:

Requirement already satisfied: pybluez in /usr/lib/python2.7/dist-packages (0.18)

Any suggestions?

closed 2019-03-18 by TziEm26

JackPoteYes, 2019-09-29:
Hi, have you been able to solve that issue ? I am in the same situation, therefore it would be helpful to share

emendir, 2023-11-29:
End of 2023 and this issue still persists.

I'm using:
- PyBluez v0.30
- Python 3.8
- Ubuntu 20.04.1
- Linux kernel 5.15.0-89-generic

yyh12138, 2024-05-19:
It's 2024 now, and this issue still persists. 
    - PyBluez v0.30
    - Python 3.11
    - Debian 12.5