pybluez/pybluez #239: Please adapt the iOS platform

lenbobopp, 2019-04-25:
@@*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:IOS
 - **Hardware:IOS internal Bluetooth adaptor
 - **Python Version:3.6.1
 - **PyBluez Version: latest

## Issue

I installed Pythonista3 on IOS device and accessed the PyBluez by StaSh. It says this platform(iOS)  is currently not supported by PyBluez 

[image]

stonehippo, 2019-04-27:
@lenbobopp Pythonista ships with the `cb` module for BLE support, which is a wrapper for the Core Bluetooth library. There’s no open developer access for Bluetooth Classic in iOS, so Pythonista can’t use it (this requires the ExternalAccessory API).

rgov, 2019-07-22:
Here's the [cb](http://omz-software.com/pythonista/docs/ios/cb.html#module-cb) module.

I'm going to close this, thanks for the answer @stonehippo.

closed 2019-07-22 by rgov