pybluez/pybluez #240: Can't install on macOS

unixb0y, 2019-05-10:
Hey guys, I have some difficulties installing pybluez on my Mac. Below is my setup and the error messages. There are 2 errors and 12 warnings, but I'll just paste the errors for now to prevent cluttering this issue ;)

## System

 - **Operating System:** macOS 10.13.6
 - **Hardware:** internal Broadcom BT chip
 - **Python Version:** 3.5.1
 - **PyBluez Version:** latest
- ** PyObjc Version:** not sure, there was an error while installing regarding 'pyobjc-framework-PhotosUI' but the bitbucket issue tracker seems not to work right now and I think it should work even without that photo thing since you won't need that for BT stuff or am I wrong?

## Issue

```
osx/_osxbt.c:676:5: error: unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
        IOBluetoothDeviceInquiryRef inquiry;
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```

```
osx/_osxbt.c:688:17: error: unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
                    IOBluetoothDeviceInquiryRef inquiryRef,
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```

jorisrymenants, 2019-05-21:
I have the same issue on macOS 10.14.4:

` gcc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/Users/jorisrymenants/anaconda/include -arch x86_64 -I/Users/jorisrymenants/anaconda/include -arch x86_64 -I/System/Library/Frameworks/IOBluetooth.framework/Headers -I/System/Library/Frameworks/CoreFoundation.framework/Headers -I/Users/jorisrymenants/anaconda/include/python3.7m -c osx/_osxbt.c -o build/temp.macosx-10.7-x86_64-3.7/osx/_osxbt.o
    osx/_osxbt.c:676:5: error: unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
        IOBluetoothDeviceInquiryRef inquiry;
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
        IOBluetoothDeviceRef
    /System/Library/Frameworks/IOBluetooth.framework/Headers/IOBluetoothUserLib.h:34:47: note: 'IOBluetoothDeviceRef' declared here
    typedef struct OpaqueIOBluetoothObjectRef *                     IOBluetoothDeviceRef;
                                                                    ^
    osx/_osxbt.c:688:17: error: unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
                    IOBluetoothDeviceInquiryRef inquiryRef,
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                    IOBluetoothDeviceRef
    /System/Library/Frameworks/IOBluetooth.framework/Headers/IOBluetoothUserLib.h:34:47: note: 'IOBluetoothDeviceRef' declared here
    typedef struct OpaqueIOBluetoothObjectRef *                     IOBluetoothDeviceRef;
                                                                    ^
    osx/_osxbt.c:703:19: warning: implicit declaration of function 'IOBluetoothDeviceInquiryCreateWithCallbackRefCon' is invalid in C99 [-Wimplicit-function-declaration]
        dd->inquiry = IOBluetoothDeviceInquiryCreateWithCallbackRefCon (&dd);
`

unixb0y, 2019-05-21:
> I have the same issue on macOS 10.14.4

Hmm, good to know. I'll try on 10.13 later today!

Rexios80, 2019-06-25:
Same problem here. Won't install and throws out a bunch of declaration errors.

unixb0y, 2019-06-25:
Only solution I found so far is checking out the commit before the „Swift 5 support“ commit.

Rexios80, 2019-06-25:
The "Swift 5 support" commit on what repo? PyObjc?

rgov, 2019-07-22:
I cannot reproduce this with Python 3.7 installed with Homebrew, and the current master:

```
% pip install -e git+https://github.com/pybluez/pybluez.git#egg=pybluez
Obtaining pybluez from git+https://github.com/pybluez/pybluez.git#egg=pybluez
  Cloning https://github.com/pybluez/pybluez.git to ./venv/src/pybluez
  Running command git clone -q https://github.com/pybluez/pybluez.git /private/tmp/foo/venv/src/pybluez
Collecting pyobjc-core>=3.1 (from pybluez)
  Downloading https://files.pythonhosted.org/packages/c8/ba/cb2f5b8a6c78de08b89f7f90cddbde3b10a8392b8350c0a0cf92120c5f43/pyobjc_core-5.2-cp37-cp37m-macosx_10_9_x86_64.whl (294kB)
     |████████████████████████████████| 296kB 1.6MB/s 
Collecting pyobjc-framework-Cocoa>=3.1 (from pybluez)
  Downloading https://files.pythonhosted.org/packages/fb/67/697f392aadf661488ef8590ffdd2a7734fa1db999a2830c136276929c58c/pyobjc_framework_Cocoa-5.2-cp37-cp37m-macosx_10_9_x86_64.whl (235kB)
     |████████████████████████████████| 235kB 4.6MB/s 
Installing collected packages: pyobjc-core, pyobjc-framework-Cocoa, pybluez
  Running setup.py develop for pybluez
Successfully installed pybluez pyobjc-core-5.2 pyobjc-framework-Cocoa-5.2
```

closed 2019-07-22 by rgov