pybluez/pybluez #243: AttributeError: 'module' object has no attribute 'setSignatureForSelector'

pokerfacett, 2019-07-12:
Traceback (most recent call last):
  File "testbluetooth.py", line 2, in <module>
    import bluetooth
  File "/usr/local/lib/python2.7/site-packages/PyBluez-0.22-py2.7.egg/bluetooth/__init__.py", line 47, in <module>
    from .macos import *
  File "/usr/local/lib/python2.7/site-packages/PyBluez-0.22-py2.7.egg/bluetooth/macos.py", line 1, in <module>
    import lightblue
  File "/usr/local/lib/python2.7/site-packages/lightblue/__init__.py", line 160, in <module>
    from _lightblue import *
  File "/usr/local/lib/python2.7/site-packages/lightblue/_lightblue.py", line 27, in <module>
    import _IOBluetooth
  File "/usr/local/lib/python2.7/site-packages/lightblue/_IOBluetooth.py", line 63, in <module>
    objc.setSignatureForSelector("IOBluetoothSDPServiceRecord",     
AttributeError: 'module' object has no attribute 'setSignatureForSelector'

pokerfacett, 2019-07-12:
please thanks!!

rgov, 2019-07-12:
Could you please use the issue template when filing bugs? You can file this again with the template.

closed 2019-07-12 by rgov