pybluez/pybluez #245: Return from HCI Command

AntoineBore, 2019-07-17:
Hi,

Is there a way to get the return from the HCI command using pybluez ?

When we execute  a command with hcitool, for example: 
`sudo hcitool cmd 0x08 0x000C 00 00`
we get a result that tell us if the command was successful or not.
```
HCI Event: 0x0e plen 4
  0E 0C 20 00
```
Another example, when you run hcidump --raw, you can get those returns:
< 01 0C 20 02 00 00
> 04 0E 04 0E 0C 20 00

Is there a way to get this using pybluez ? The only return from bluez.hci_send_cmd(sock, OGF_LE_CTL, service, cmd_pkt) seems to be 0 and I cannot find a similar function that give me this result.

rgov, 2019-07-22:
The Python function just calls through to the Bluez function of the same name:

https://github.com/pybluez/pybluez/blob/9d52f97ca052757edbd6683e72ee2f54d58cbc1b/bluez/btmodule.c#L2038

If we look at how `hcitool` does it here: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/tools/hcitool.c#n1165

It looks like you are supposed to read/recv from the socket. Please let me know if that works.

AntoineBore, 2019-07-23:
Thank you for your response. I have managed to do what I wanted:
When you send a command with "hci_send_cmd", you can read the response from the Bluetooth module on the Bluetooth socket.

The event name of this response is "EVT_CMD_COMPLETE".

rgov, 2019-07-23:
Great! Thanks.

closed 2019-07-23 by rgov