pybluez/pybluez #246: error communicating with local bluetooth adapter

bieniu, 2019-07-22:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** Raspbian Stretch
 - **Hardware:** Raspberry Pi 3 internal adapter
 - **Python Version:** 3.7.2
 - **PyBluez Version:** 0.22

## Issue

Sometimes during scan I have this error:
```
2019-07-22 18:19:11 ERROR (SyncWorker_4) [homeassistant.components.bluetooth_tracker.device_tracker] Error looking up Bluetooth device
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/bluetooth/bluez.py", line 31, in discover_devices
    lookup_class=lookup_class, device_id=device_id)
_bluetooth.error: (110, 'Connection timed out')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/bluetooth_tracker/device_tracker.py", line 102, in update_bluetooth_once
    for dev in discover_devices():
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/bluetooth_tracker/device_tracker.py", line 57, in discover_devices
    lookup_class=False, device_id=device_id)
  File "/srv/homeassistant/lib/python3.7/site-packages/bluetooth/bluez.py", line 34, in discover_devices
    raise BluetoothError ("error communicating with local "
bluetooth.btcommon.BluetoothError: error communicating with local bluetooth adapter
```
After the error, the scan doesn't see any devices. It is necessary to reset the adapter to scan devices again:
```
sudo hciconfig hci0 reset
```

rgov, 2019-07-22:
Are you sure this is an issue with PyBluez? It looks like PyBluez itself can't talk to the Bluetooth controller. Are other Bluetooth functions working?

bieniu, 2019-07-23:
I'm not 100% sure but after the error when I try to pair a device from the Raspbian desktop, system displays a list of available BT devices in the area - BT works.

rgov, 2019-07-23:
Are you using the latest code installed from this repo? I notice that the error comes from the `hci_inquiry()` function failing. But the error message I see in the latest version of the code ends with a colon followed by a reason. Not sure if you are using older code before that was added.

This is probably below PyBluez, though.

bieniu, 2019-07-23:
@rgov I'm using 0.22 release installed via pip.

rgov, 2019-07-23:
Try `pip install -e git+https://github.com/pybluez/pybluez.git#egg=pybluez` and see if it still has issues.

bieniu, 2019-07-23:
I will try, thanks.

bieniu, 2019-07-27:
Issue is kernel related. All kernels 4.19.x to Raspberry Pi have this issue. Fix will be available in kernel 4.19.58.

closed 2019-07-27 by bieniu

pgilad, 2019-08-18:
@bieniu I'm seeing this behavior under kernel `4.19.58`:
```
(venv) pi@Camelot:~/bluetooth-test $ uname -a
Linux Camelot 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux
```

I'm running a simple script:
```
import bluetooth
nearby_devices = bluetooth.discover_devices(
            duration=8,
            lookup_names=True,
            flush_cache=True,
            lookup_class=False,
            device_id=-1,
        )
```
And it hangs on this line. I've installed release `0.22` and manually from @rgov command and the process gets stuck on both runs (unless I exit it).

It seems that running `sudo hciconfig hci0 reset` does solve the problem, but only temporarily. This seems to be a new issue, I'm not sure whether it's actually a kernel or `pybluez` issue though.. Running `bluetoothctl` and scanning runs without any issues.