pybluez/pybluez #247: Is there a plan to provide an API to set the advertisement interval time

YanMinge, 2019-07-23:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**
       Linux raspberry 4.19.57-v7+
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
       Raspberry Pi 3 Model B+
 - **Python Version:**
       python 3.7
 - **PyBluez Version:**
       bluez-5.50

## Issue
Similar to these issue。
https://stackoverflow.com/questions/50996419/bluez-example-advertisement-interval-linux
http://www.itkeyword.com/doc/9107174144006173442/is-there-a-way-to-increase-ble-advertisement-frequency-in-bluez

When developing the GATT service, I needed a host to automatically connect to the Raspberry Pi GATT service. I hope that the time of this connection is fast enough. but the advertisement time can‘t be set. In my test, the average interval time of advertisement takes more than 2s，I really need to be able to set this time, do I have some ways to change the adv. interval time ?
The time for this setting should be an integer multiple of 0.625ms.

I hope to hear from you.

rgov, 2019-07-23:
If you can find a way to do this with the Bluez C API it should be possible to do with PyBluez.

YanMinge, 2019-07-23:
In bluez 5.5.0  `emulator/le.c` it have an function named `cmd_le_set_adv_parameters`
and in `tools/ibeacon.c`, it also can use `set_adv_parameters` to set the adv. interval time. but for gatt server or advertising, I haven't found same function.

closed 2019-07-24 by YanMinge

cagnulein, 2020-09-09:
have a look here https://github.com/colin-guyon/py-bluetooth-utils/blob/master/bluetooth_utils.py