pybluez/pybluez #248: OSerror bt.connect msbt

111-Sj-111, 2019-07-23:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**
Windows 10 64bits 
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)

 - **Python Version:**
Python 3.7
 - **PyBluez Version:**
0.22

## Issue

Whenever I try to connect my wiiboard to my pc to make a manual acquisition from it, I get this error message and I don't know why 👍 
SO that i can retrieve data from all sensors of the Wiiboard, I used the runExperiment.py script from this WiiSway github : https://github.com/CamMerrill/WiiSway (don't worry, I modified it as needed to get to this point)
```
Traceback (most recent call last):
  File "runExperiment.py", line 95, in <module>
    main()
  File "runExperiment.py", line 28, in main
    board.connect1(address) #The wii board must be in sync mode at this time
  File "C:\Users\Smartcheck\Documents\WiiSway-master\wiiboard.py", line 95, in connect 
    self.receivesocket.connect(("00:26:59:76:28:0F",0x13))
  File "C:\Users\Smartcheck\Documents\WiiSway-master\bluetooth\msbt.py", line 93, in connect
    bt.connect (self._sockfd, addr, port)
OSError
```

Would anyone know where this error could come from ?

Kind Regards,

Please provide information necessary to reproduce the issue, including sample
code.
python runExperiment.py 

If you have long logs, please post them on https://gist.github.com and link to
the Gist here.

rgov, 2019-07-23:
Can you first try to get PyBluez examples working on your system, so we can confirm whether the issue is with WiiSway or not?

111-Sj-111, 2019-07-23:
Hey, of course, here you go :

```
C:\Users\Smartcheck\Documents\WiiSway-master>python testrfcomm.py 00:26:59:76:28:0F
Searching for SampleServer on 00:26:59:76:28:0F
connecting to "None" on 00:26:59:76:28:0F
Traceback (most recent call last):
  File "testrfcomm.py", line 43, in <module>
    sock.connect((host, port))
  File "C:\Users\Smartcheck\Documents\WiiSway-master\bluetooth\msbt.py", line 93, in connect
    bt.connect (self._sockfd, addr, port)
OSError
```

111-Sj-111, 2019-07-24:
I've got the same type of error with the `bt.listen` function from the msbt.py script when I try to run the l2cap-server.py script.

beadysea, 2019-07-25:
I ran l2capserver.py and got:
```
Traceback (most recent call last):
  File "l2capserver.py", line 7, in <module>
    server_sock=bluetooth.BluetoothSocket( bluetooth.L2CAP )
  File "C:\Users\<user>\AppData\Local\Programs\Python\Python37\lib\site-packages\pybluez-0.22-py3.7-win-amd64.egg\bluetooth\msbt.py", line 50, in __init__
ValueError: invalid protocol
```

beadysea, 2019-07-25:
If you are trying to use L2CAP I'm afraid your out of luck at the moment. After some investigating it appears access to the L2CAP layer happens in kernel mode on windows. Gaining access to L2CAP would require a driver so unless anyone is up for writing a driver I guess we are stuck with RFCOMM server and client on Windows.

closed 2019-07-29 by beadysea