pybluez/pybluez #249: Set up Read the Docs

rgov, 2019-07-23:
Tagging @beadysea here.

Read the Docs cannot generate documentation from this repo. Here is the error log:

https://readthedocs.org/projects/pybluez/builds/9413084/

It expects a file called `contents.rst`.

beadysea, 2019-07-23:
Thanks, I'm on it.

rgov, 2019-07-23:
Weird, the content isn't visible at https://pybluez.readthedocs.io/en/latest/ but the PDF looks good: https://buildmedia.readthedocs.org/media/pdf/pybluez/latest/pybluez.pdf

rgov, 2019-07-23:
Just caching issues maybe. I think it works!

closed 2019-07-23 by rgov

beadysea, 2019-07-23:
It works, did an F5 refresh on my browser.

beadysea, 2019-07-23:
@rgov can you update the doc link on the repo from https://pybluez.github.io/  to https://pybluez.readthedocs.io/en/latest/