pybluez/pybluez #261: Can we drop support for Python versions <= 2.6?

cclauss, 2019-07-29:
Testing on Python 2.7, 3.5, 3.6, and 3.7 is reasonable but https://snarky.ca/stop-using-python-2-6 was written on 2015.

*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:**
 - **PyBluez Version:**

## Issue

Please provide information necessary to reproduce the issue, including sample
code.

If you have long logs, please post them on https://gist.github.com and link to
the Gist here.

rgov, 2019-07-29:
What does Ubuntu LTS 14.04 ship with? Are there usage stats?

macOS has shipped Python 2.7 forever.

cclauss, 2019-07-29:
Ubuntu LTS 14.04 ship with Python 2.7 and 3.4.

$ __pypistats python_major PyBluez__
```
| category | percent | downloads |
|----------|--------:|----------:|
|        3 |  60.46% |    34,833 |
|        2 |  31.39% |    18,082 |
| null     |   8.15% |     4,695 |
| Total    |         |    57,610 |
```
$ __pypistats python_minor PyBluez__
```
| category | percent | downloads |
|----------|--------:|----------:|
|      2.7 |  31.38% |    18,077 |
|      3.7 |  24.87% |    14,328 |
|      3.6 |  16.80% |     9,680 |
|      3.5 |  13.54% |     7,799 |
| null     |   8.15% |     4,695 |
|      3.4 |   5.09% |     2,930 |
|      3.8 |   0.15% |        89 |
|      3.3 |   0.01% |         7 |
|      2.6 |   0.01% |         5 |
| Total    |         |    57,610 |
```
$ __pypistats system PyBluez__
```
| category | percent | downloads |
|----------|--------:|----------:|
| Linux    |  67.23% |    38,732 |
| null     |  13.15% |     7,575 |
| Windows  |  12.80% |     7,376 |
| Darwin   |   6.75% |     3,891 |
| other    |   0.06% |        36 |
| Total    |         |    57,610 |
```
@hugovk

rgov, 2019-07-29:
Yes, I'm fine with dropping 2.6 and lower.

hugovk, 2019-07-29:
As it happens, Ubuntu LTS 14.04 also went EOL in April 2019.

https://linuxlifecycle.com

cclauss, 2019-07-29:
__End of Standard Support__ != __End of Life__ -- https://wiki.ubuntu.com/Releases

hugovk, 2019-07-29:
By the way, if you paste a table directly without a code block, it'll render it as a Markdown table:

| category | percent | downloads |
|----------|--------:|----------:|
|      2.7 |  31.38% |    18,077 |
|      3.7 |  24.87% |    14,328 |
|      3.6 |  16.80% |     9,680 |
|      3.5 |  13.54% |     7,799 |
| null     |   8.15% |     4,695 |
|      3.4 |   5.09% |     2,930 |
|      3.8 |   0.15% |        89 |
|      3.3 |   0.01% |         7 |
|      2.6 |   0.01% |         5 |
| Total    |         |    57,610 |

cclauss, 2019-07-29:
Remove backtabs in tables above.

rgov, 2019-12-28:
Marking as a dupe of #266; we're going to drop Python 2.x.

closed 2019-12-28 by rgov