pybluez/pybluez #263: Contrib is not Python 3 compatible

cclauss, 2019-07-30:
Discovered in #254 - https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test/sap_client.py contains legacy __print__ statements that need to be converted to __print()__ function to be compatible with Python 3.

*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:**
 - **PyBluez Version:**

## Issue

Please provide information necessary to reproduce the issue, including sample
code.

If you have long logs, please post them on https://gist.github.com and link to
the Gist here.

rgov, 2019-07-30:
Hm, this needs to be filed against the upstream Bluez project.

rgov, 2019-08-29:
Closing this because I don't think we control it.

closed 2019-08-29 by rgov