pybluez/pybluez #266: End python 2 support

beadysea, 2019-08-13:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## Systems Affected

 - **Operating System:** All
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:** <= 2.7
 - **PyBluez Version:** > 0.22

## Issue/ Request for comment

I'm posting this issue because Python 2 reaches its end of life on 1st January 2020 [PEP373](https://www.python.org/dev/peps/pep-0373/) and we need to map out a plan for dropping Python 2 support. 
Some transitioning work has started with #261, #257, #258, #259, #260 by @cclauss and #262 by @beadysea. At present we are still supporting and testing against Python 2.7.

Python 2.7 accounts for almost 32% of Pybluez downloads. 

The question then is what would be a realistic timescale for dropping Python 2.7 support?

rgov, 2019-08-13:
How's this?

1. Post a new release to PyPI, the last to support Python 2.7.
2. Remove Python 2.7 from our testing matrix and from the documentation.
3. Make a major or minor version number bump.
4. Changes going forward are not required to work with Python 2.7. Code style should reflect a Python 3 codebase.

beadysea, 2019-08-13:
I was thinking along the lines of v 0.23 - last python2 version, next release v 0.30 - python 3

rgov, 2019-12-28:
We have released 0.23, so we can discontinue Python 2 support.

rgov, 2019-12-31:
I've made initial progress towards this, and I will accept pull requests that take this further.

closed 2019-12-31 by rgov

hugovk, 2019-12-31:
Please see PR https://github.com/pybluez/pybluez/pull/295.