pybluez/pybluez #267: Binding a server socket in MacOS fails to retrieve an available service channel

phoenix-meadowlark, 2019-08-14:
## System

 - **Operating System**: MacOS `10.14.6`
 - **Hardware**: N/A
 - **Python Version**: Python `3.7.4`
 - **PyBluez Version**: Installed 8/14/2019 via `python3 -m pip install git+https://github.com/pybluez/pybluez.git`
- **Installation of LightAquaBlue**:
```
$ sudo xcode-select -s /Applications/Xcode.app/Contents/Developer
$ sudo python3 setup.py install

# Line below to address the following error:
# ImportError: Cannot load LightAquaBlue framework, not found at /usr/local/lib/python3.7/site-packages/lightblue/LightAquaBlue.framework
$ ln -s /Library/Frameworks/LightAquaBlue.framework /usr/local/lib/python3.7/site-packages/lightblue/LightAquaBlue.framework
```

## Issue
Running the following 
```python
import bluetooth
server_socket=bluetooth.BluetoothSocket()
port = 0 # must be zero for MacOS
server_socket.bind(("",port))
```
results the following error:
```python
Traceback (most recent call last):
  File "bluetooth_tests.py", line 4, in <module>
    server_socket.bind(("",port))
  File "/usr/local/lib/python3.7/site-packages/bluetooth/macos.py", line 110, in bind
    return self._sock.bind(addrport)
  File "/usr/local/lib/python3.7/site-packages/lightblue/_bluetoothsockets.py", line 273, in bind
    address = (address[0], _getavailableport(self.__conn.proto))
  File "/usr/local/lib/python3.7/site-packages/lightblue/_bluetoothsockets.py", line 69, in _getavailableport
    "Could not retrieve an available service channel")
lightblue._lightbluecommon.BluetoothError: [Errno -536870208] Could not retrieve an available service channel
```

which is caused by [this line](https://github.com/pybluez/pybluez/blob/adbbbdd7058b6adaae06eec7e95f3f82cf2ae09f/macos/_bluetoothsockets.py#L66)
```python
result = BBServiceAdvertiser.removeService_(servicerecordhandle)
```
returning `-536870208` instead of `_macutil.kIOReturnSuccess`, which takes a value of `0`.

Looking into ObjectiveC backend, [`BBServiceAdvertiser._removeService`](https://github.com/pybluez/pybluez/blob/adbbbdd7058b6adaae06eec7e95f3f82cf2ae09f/macos/LightAquaBlue/BBServiceAdvertiser.m#L153) calls `IOBluetoothRemoveServiceWithRecordHandle`. (I don't know much about ObjectiveC or Mac system programing, but [this](http://codeworkshop.net/objc-diff/sdkdiffs/macos/10.15/IOBluetooth.html) implies it will be removed soon).
```ObjectiveC
+ (IOReturn)removeService:(BluetoothSDPServiceRecordHandle)handle
{
    // TODO: We should switch to using [IOBluetoothSDPServiceRecord removeServiceRecord]
    // but we don't know how to get an IOBluetoothSDPServiceRecord instance from a handle.
	return IOBluetoothRemoveServiceWithRecordHandle(handle);
}
```

I don't have the background to say what the cause of this is, but I would greatly appreciate any help.

wenchao1020, 2019-09-04:
open bluetooth before you run

closed 2019-10-11 by stale[bot]

AInitikesh, 2019-11-06:
I am also facing the same issue. Any solution ?

phoenix-meadowlark, 2019-11-07:
I didn't find any. I couldn't find any python bluetooth library that functioned consistently enough cross-platform, so I conceded to using noblejs. It wouldn't be hard to wrap noblejs in a server that python communicates to.

Or, maybe you could try opening bluetooth before you run. <sarcasm/>

rgov, 2019-11-08:
PyQt5 supports BTLE on multiple platforms. You could look at that instead.

-536870208 in hexadecimal is 0xE00002C0 which is an IOKit error. [Some docs on IOKit errors.](http://mirror.informatimago.com/next/developer.apple.com/qa/qa2001/qa1075.html) From this we have system = 0x38 (`sys_iokit`), subsystem = 0x00 (`sub_iokit_common`), code = 0x2C0 (`kIOReturnNoDevice`). See [IOReturn.h](https://opensource.apple.com/source/xnu/xnu-4903.241.1/iokit/IOKit/IOReturn.h).

ozgurgundogan, 2019-12-13:
Ive had the same problem. What is the solution then ?

Yamyee, 2020-12-27:
you can update Objective-C code cause the api was depreciated，then build the Objective-C code with xcode and find the framework ,copy it to the lightblue directory.