pybluez/pybluez #268: Read the Docs import error

beadysea, 2019-08-15:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** Linux
 - **Hardware:** N/A
 - **Python Version:** 2.7
 - **PyBluez Version:** Latest

## Issue
Read the Docs is failing to import functions which have their `__doc__` defined in `__init__.py`.
See build log https://readthedocs.org/projects/pybluez/builds/9515710/

**Example from build log**
```
WARNING: autodoc: failed to import function u'advertise_service' from module u'bluetooth'; the following exception was raised:
Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/pybluez/envs/latest/lib/python2.7/site-packages/sphinx/ext/autodoc/importer.py", line 154, in import_module
    __import__(modname)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pybluez/checkouts/latest/bluetooth/__init__.py", line 282, in <module>
    """
AttributeError: attribute '__doc__' of 'instancemethod' objects is not writable
```

rgov, 2019-08-29:
I think we need to switch Read the Docs to use Python 3 instead of 2.7. That might resolve it.

beadysea, 2019-09-16:
Switched Read the Docs to use Python 3, seems to have resolved the issue.

closed 2019-09-16 by beadysea

tomitrescak, 2021-05-11:
Hello. Is there any way now to make this work with Python 2.7?

lavanyascyberlife, 2022-02-21:
if you using pybluez 0.23. You need to run under the python 3.0 environment.
if you want to import Bluetooth in python 2.7. You need to install pybluez 0.22 instead of 0.23.

sudo python2.7 -m pip install pybluez="0.22" will solve for python 2.7

userse31, 2022-12-29:
The decision to split python into 2.7 and 3+ has got to be one of the biggest mistakes in all of history. So ANNOYING!