pybluez/pybluez #271: Switch to GitHub Actions for CI

rgov, 2019-08-29:
Travis is basically dead, and GitHub has introduced its own CI offering, which is currently in beta.

We should be able to produce builds for different systems (Mac, Windows, and Linux) and more easily turn out releases for PyPI.

rgov, 2019-08-29:
I have this prototyped over on my fork, but until the PyBluez org gets invited to the beta, we cannot use it.

rgov, 2019-08-29:
- [ ] Fix Windows build 

setup.py cannot find Visual Studio.

beadysea, 2019-09-16:
@rgov see [Windows Compilers for Python](https://wiki.python.org/moin/WindowsCompilers).

rgov, 2019-09-16:
Thanks, interesting. The Windows Server 2016 R2 environment has Visual Studio 2017 Enterprise version VisualStudio/15.9.15+28307.812 installed. The Windows Server 2019 environment has Visual Studio 2019 Enterprise version VisualStudio/16.1.6+29102.190 installed. (What a mouthful.)

This is next to useless to me to figure out what version of Visual C++ these include but [Wikipedia provides some info](https://en.wikipedia.org/wiki/Microsoft_Visual_C%2B%2B#Internal_version_numbering).

Therefore, I _think_, that with the Windows 2016 environment, we have MSVC++ 14.16 and with the 2019 environment we have MSVC++ 14.21.

Which means we can only build for Python 3.5+ on GitHub Actions. However, we could install MinGW for earlier releases of Python. Otherwise, installing the Microsoft compiler seems like a headache.

beadysea, 2019-09-16:
Just had a quick look here: https://help.github.com/en/articles/software-in-virtual-environments-for-github-actions#mingw.
As there is an environment path to mingw/bin I'd guess that MinGW 8.1.0 is already installed on the Windows 2016 and 2019 environments. `setup.py` would need to be modified to detect this path and use mingw when a suitable msvc compiler is not found.

rgov, 2019-11-18:
I got a response from GitHub's ethomson [in this thread](https://github.community/t5/GitHub-Actions/Locating-the-Visual-Studio-SDK-for-Python-build/m-p/35945#M2384):

> We install the Visual Studio build tooling, so that you can use the compilers, linkers, etc.  However, we do not install the Windows SDK.
 
> If you want to install one of the Windows SDKs, the easiest way to do that is to install it with chocolatey.  For example, you can install the https://chocolatey.org/packages/windows-sdk-10.0 package and I think that your build script will be able to locate it.

rgov, 2019-12-28:
This was fixed in #292 with help from @Arusekk.

closed 2019-12-28 by rgov