pybluez/pybluez #277: Using in Arch / Manjaro

simonb02, 2019-10-14:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** Manjaro(Linux)
 - **Hardware:** Thankpad L460
 - **Python Version:** 3.7
 - **PyBluez Version:**

## Issue

Hi,
I'm having an issue with libboost_python lib. I have installed that lib but it is named different.
It should be fixable by using this:
` if platform.system() == 'Linux': 
     # boost_python-py34 
     boost_lib = 'boost_python-py'+str(python_version[0])+str(python_version[1])`
Maybe someone can help me fixing this issue. :-) 
With regards
Simon

rgov, 2019-10-14:
It sounds like you already have the solution. If you have installed PyBluez and need to edit the file, you can do `import pybluez; print(pybluez.__path__)` which will tell you where the files are installed.

If you would like to make PyBluez capable of detecting the library on your system, I'll accept a pull request, but it has to be careful not to impair any other system.

simonb02, 2019-10-16:
Thanks a lot. I found a way how not to influence other distributions:
```
if platform.dist()[0] == 'Arch'
boost_lib = 'boost_python-py'+str(python_version[0])+str(python_version[1])
```

simonb02, 2019-10-16:
But I can't find the file where you added this lib. By reading the setup.py I thought it must be in the folder Bluetooth, but I can't find a file in the the folder, which does it.

closed 2019-11-22 by stale[bot]