pybluez/pybluez #279: OSError when program try to use functions from bluetooth._msbt

liiliiliil, 2019-10-24:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** Windows 10 64bits
 - **Hardware:** internal Bluetooth adaptor
 - **Python Version:** 3.7.4
 - **PyBluez Version:** 0.22
 - 

## Issue

Program raised an OSError with no other information whenever it trys to call functions from bluetooth._msbt. Like the code below, 
`import bluetooth

nearby_devices = bluetooth.discover_devices(lookup_names=True)
for addr, name in nearby_devices:
    print("  %s - %s" % (addr, name))
 
    services = bluetooth.find_service(address=addr)
    for svc in services:
        print("Service Name: %s"    % svc["name"])
        print("    Host:        %s" % svc["host"])
        print("    Description: %s" % svc["description"])
        print("    Provided By: %s" % svc["provider"])
        print("    Protocol:    %s" % svc["protocol"])
        print("    channel/PSM: %s" % svc["port"])
        print("    svc classes: %s "% svc["service-classes"])
        print("    profiles:    %s "% svc["profiles"])
        print("    service id:  %s "% svc["service-id"])
        print("")`

I got
` xx:xx:xx:xx:xx:xx - devicename
Traceback (most recent call last):
  File "e:/Project/BMKSim_PC_Part/Python/Bluetooth_test2.py", line 7, in <module>
    services = bluetooth.find_service(address=addr)
  File "E:\Miniconda3\envs\exp_py37\lib\site-packages\pybluez-0.22-py3.7-win-amd64.egg\bluetooth\msbt.py", line 231, in find_service
OSError`

In other cases, like
`import bluetooth

bd_addr = "xx:xx:xx:xx:xx:xx"

port = 11

sock=bluetooth.BluetoothSocket( bluetooth.RFCOMM )
sock.connect((bd_addr, port))

sock.send("hello!!")

sock.close()`

I got 
`Traceback (most recent call last):
  File "e:/Project/BMKSim_PC_Part/Python/Bluetooth_test1.py", line 8, in <module>
    sock.connect((bd_addr, port))
  File "E:\Miniconda3\envs\exp_py37\lib\site-packages\pybluez-0.22-py3.7-win-amd64.egg\bluetooth\msbt.py", line 84, in connect
OSError`

Functions like bluetooth.advertise_service( server_sock, "FooBar Service", uuid ) and bluetooth.find_service( uuid = uuid ) also raised OSError like that.

Can anyone help me with this problem? Thanks a lot!

liiliiliil, 2019-10-24:
I found a solution, i.e. uninstall this module and install another one [PyBluez-win10](https://pypi.org/project/PyBluez-win10/). If you have all pre-requirements, just simply enter `pip install PyBluez-win10`  in your cmd and you will get pybluez-win10. So I think there is a compatibility problem if you directly install this module in Windows 10. Thanks Kiven a lot for your PyBluez-win10! :)

closed 2019-10-24 by liiliiliil

netokey, 2020-11-20:
> I found a solution, i.e. uninstall this module and install another one [PyBluez-win10](https://pypi.org/project/PyBluez-win10/). If you have all pre-requirements, just simply enter `pip install [...]

PyBluez-win10 is not available now. Any alternative?

fbouchet, 2020-11-26:
@netokey : in my case, installing the previous version of pybluez with `pip install PyBluez==0.22` fixed the current issue

lshaf, 2021-02-08:
any workaround for python 3.7 ?

welkensoyo, 2022-08-11:
pip install git+https://github.com/pybluez/pybluez.git#egg=pybluez