pybluez/pybluez #285: Cannot install on macOS Catalina

seanngpack, 2019-11-23:
## System

 - **Operating System:** macOS Catalina
 - **Hardware:** MBP 15
 - **Python Version:** 3.7.4
 - **PyBluez Version:** latest

## Issue

I installed PyObjC, now onto `pip install pybluez` it cannot install throwing this error:

`Command ""/Users/seanngpack/Programming Stuff/Projects/swag-scanner/venv/bin/python3" -u -c "import setuptools, tokenize;__file__='/private/var/folders/hd/gppgkbjn16j86w5yj954ls500000gn/T/pip-install-n_42y6p_/pybluez/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/var/folders/hd/gppgkbjn16j86w5yj954ls500000gn/T/pip-record-4tbv7v6o/install-record.txt --single-version-externally-managed --compile --install-headers "/Users/seanngpack/Programming Stuff/Projects/swag-scanner/venv/include/site/python3.7/pybluez"" failed with error code 1 in /private/var/folders/hd/gppgkbjn16j86w5yj954ls500000gn/T/pip-install-n_42y6p_/pybluez/`

Also tried other peoples' forks and different errors occurred. Anyone else having these issues?

rgov, 2019-11-23:
Your initial report was not a full log and doesn't say what failed, so it cannot be diagnosed, but I'm glad that using the repo's latest code worked for you.

Your update isn't clear to me where this is failing—are you running a particular example?

`gattlib` is only used for BTLE, which is not supported by PyBluez on macOS. You might have luck with using PyQt5's Bluetooth APIs.

VolodymyrBiryuk, 2019-11-28:
Can't install on MacOS Catalina either:
- Hardware: MBP 15
- Python Version: 3.6.9
- PyBluez Version: latest

```console
Building wheels for collected packages: pybluez
  Building wheel for pybluez (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/xxx/xxx/xxx/xxx/xxx/venv/bin/python3.6 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/57/lny8x6l561131gcx79zms3fr0000gn/T/pip-install-ivlo40kf/pybluez/setup.py'"'"'; __file__='"'"'/private/var/folders/57/lny8x6l561131gcx79zms3fr0000gn/T/pip-install-ivlo40kf/pybluez/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/57/lny8x6l561131gcx79zms3fr0000gn/T/pip-wheel-z8_maq_x --python-tag cp36
       cwd: /private/var/folders/57/lny8x6l561131gcx79zms3fr0000gn/T/pip-install-ivlo40kf/pybluez/
  Complete output (55 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.15-x86_64-3.6
  creating build/lib.macosx-10.15-x86_64-3.6/bluetooth
  copying bluetooth/btcommon.py -> build/lib.macosx-10.15-x86_64-3.6/bluetooth
  copying bluetooth/ble.py -> build/lib.macosx-10.15-x86_64-3.6/bluetooth
  copying bluetooth/__init__.py -> build/lib.macosx-10.15-x86_64-3.6/bluetooth
  copying bluetooth/bluez.py -> build/lib.macosx-10.15-x86_64-3.6/bluetooth
  copying bluetooth/msbt.py -> build/lib.macosx-10.15-x86_64-3.6/bluetooth
  copying bluetooth/widcomm.py -> build/lib.macosx-10.15-x86_64-3.6/bluetooth
  copying bluetooth/osx.py -> build/lib.macosx-10.15-x86_64-3.6/bluetooth
  running build_ext
  building 'bluetooth._osxbt' extension
  creating build/temp.macosx-10.15-x86_64-3.6
  creating build/temp.macosx-10.15-x86_64-3.6/osx
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include -I/System/Library/Frameworks/IOBluetooth.framework/Headers -I/System/Library/Frameworks/CoreFoundation.framework/Headers -I/Users/volodymyrbiryuk/.pyenv/versions/3.6.9/include/python3.6m -c osx/_osxbt.c -o build/temp.macosx-10.15-x86_64-3.6/osx/_osxbt.o
  osx/_osxbt.c:676:5: error: unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
      IOBluetoothDeviceInquiryRef inquiry;
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      IOBluetoothDeviceRef
  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/IOBluetooth.framework/Headers/IOBluetoothUserLib.h:34:47: note: 'IOBluetoothDeviceRef' declared here
  typedef struct OpaqueIOBluetoothObjectRef *                     IOBluetoothDeviceRef;
                                                                  ^
  osx/_osxbt.c:688:17: error: unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
                  IOBluetoothDeviceInquiryRef inquiryRef,
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                  IOBluetoothDeviceRef
  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/IOBluetooth.framework/Headers/IOBluetoothUserLib.h:34:47: note: 'IOBluetoothDeviceRef' declared here
  typedef struct OpaqueIOBluetoothObjectRef *                     IOBluetoothDeviceRef;
                                                                  ^
  osx/_osxbt.c:703:19: warning: implicit declaration of function 'IOBluetoothDeviceInquiryCreateWithCallbackRefCon' is invalid in C99 [-Wimplicit-function-declaration]
      dd->inquiry = IOBluetoothDeviceInquiryCreateWithCallbackRefCon (&dd);
                    ^
  osx/_osxbt.c:703:17: warning: incompatible integer to pointer conversion assigning to 'IOBluetoothDeviceRef' (aka 'struct OpaqueIOBluetoothObjectRef *') from 'int' [-Wint-conversion]
      dd->inquiry = IOBluetoothDeviceInquiryCreateWithCallbackRefCon (&dd);
                  ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  osx/_osxbt.c:705:5: warning: implicit declaration of function 'IOBluetoothDeviceInquirySetCompleteCallback' is invalid in C99 [-Wimplicit-function-declaration]
      IOBluetoothDeviceInquirySetCompleteCallback (dd->inquiry,
      ^
  osx/_osxbt.c:717:5: warning: implicit declaration of function 'IOBluetoothDeviceInquiryStart' is invalid in C99 [-Wimplicit-function-declaration]
      IOBluetoothDeviceInquiryStart (dd->inquiry);
      ^
  osx/_osxbt.c:721:5: warning: implicit declaration of function 'IOBluetoothDeviceInquiryDelete' is invalid in C99 [-Wimplicit-function-declaration]
      IOBluetoothDeviceInquiryDelete (dd->inquiry);
      ^
  osx/_osxbt.c:774:9: warning: implicit declaration of function 'Py_InitModule3' is invalid in C99 [-Wimplicit-function-declaration]
      m = Py_InitModule3("_osxbt", osxbt_methods, osxbt_doc);
          ^
  osx/_osxbt.c:774:7: warning: incompatible integer to pointer conversion assigning to 'PyObject *' (aka 'struct _object *') from 'int' [-Wint-conversion]
      m = Py_InitModule3("_osxbt", osxbt_methods, osxbt_doc);
        ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  7 warnings and 2 errors generated.
  error: command 'clang' failed with exit status 1
[... cut ...] tall-ivlo40kf/pybluez/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/57/lny8x6l561131gcx79zms3fr0000gn/T/pip-record-gfpmxc7i/install-record.txt --single-version-externally-managed --compile --install-headers /Users/volodymyrbiryuk/git/mast/enviment/test/venv/include/site/python3.6/pybluez Check the logs for full command output.

```

hbg, 2019-12-11:
Me neither...

rgov, 2019-12-11:
Are you installing from the Git repository, rather than from PyPI?

ranxuebin, 2019-12-15:
```
unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
        IOBluetoothDeviceInquiryRef inquiry;
```
neither.

marians, 2019-12-15:
Same here, tried `pip install PyBluez`.

```
Collecting PyBluez
  Using cached https://files.pythonhosted.org/packages/c1/98/3149481d508bee174335be6725880f00d297afebe75c15e917af8f6fe169/PyBluez-0.22.zip
Building wheels for collected packages: PyBluez
...

clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -I/System/Library/Frameworks/IOBluetooth.framework/Headers -I/System/Library/Frameworks/CoreFoundation.framework/Headers -I/usr/local/Cellar/python/3.7.5/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c osx/_osxbt.c -o build/temp.macosx-10.15-x86_64-3.7/osx/_osxbt.o
  osx/_osxbt.c:676:5: error: unknown type name 'IOBluetoothDeviceInquiryRef'; did you mean 'IOBluetoothDeviceRef'?
...

rgov, 2019-12-15:
Install from the GitHub repo and not from the last published release on PyPI. If it doesn’t work please comment. 

We’d like to release a new version to PyPI but need a volunteer to help with Windows 10.

vrai-univpm, 2019-12-23:
Good morning I try to install pybluez on macOS Catalina if I use PyPI package the installation finish successfully. When I try to use it with some example I have this error on the ` import bluetooth`
How can I solve this problem?
I report the error log:
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/bluetooth/__init__.py", line 47, in <module>
    from .osx import *
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/bluetooth/osx.py", line 1, in <module>
    import lightblue
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/lightblue/__init__.py", line 160, in <module>
    from ._lightblue import *
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/lightblue/_lightblue.py", line 29, in <module>
    from . import _LightAquaBlue
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/lightblue/_LightAquaBlue.py", line 33, in <module>
    _FRAMEWORK_PATH)
ImportError: Cannot load LightAquaBlue framework, not found at /Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/lightblue/LightAquaBlue.framework`

Eggwise, 2019-12-24:
Same as @vrai-univpm getting this error:

```
  File "/Library/Python/3.7/site-packages/bluetooth/__init__.py", line 47, in <module>
    from .macos import *
  File "/Library/Python/3.7/site-packages/bluetooth/macos.py", line 1, in <module>
    import lightblue
  File "/Library/Python/3.7/site-packages/lightblue/__init__.py", line 160, in <module>
    from ._lightblue import *
  File "/Library/Python/3.7/site-packages/lightblue/_lightblue.py", line 29, in <module>
    from . import _LightAquaBlue
  File "/Library/Python/3.7/site-packages/lightblue/_LightAquaBlue.py", line 33, in <module>
    _FRAMEWORK_PATH)
ImportError: Cannot load LightAquaBlue framework, not found at /Library/Python/3.7/site-packages/lightblue/LightAquaBlue.framework
```

Using MacOS Catalina. 10.15.2. Retina mid 2015
Python 3

rgov, 2019-12-24:
Come on, are you all installing from PyPI when I explicitly just said in the last comment NOT to do that? And you expect it to work?

Try:

```
pip install git+https://github.com/pybluez/pybluez.git
```

If it still doesn't work, then we can investigate why not.

closed 2019-12-24 by rgov

otger, 2020-12-21:
> First, don't install from PyPI, it is outdated, as I said in my previous comment.
> 
> ```
> pip uninstall pybluez
> pip install git+https://github.com/pybluez/pybluez.git
> ```
> 
> You may try to `import bluetooth` and get the above error if LightAquaBlue.framework isn't built as part of the install process.

I tried this but it fails when building the wheel for PyBluez and then it installs from setup.py. After installing from setup.py, it fails when trying to import Bluetooth:
`ImportError: Cannot load LightAquaBlue framework, not found at /Users/otger/development/virgo/virgo_venv/lib/python3.8/site-packages/lightblue/LightAquaBlue.framework
`

MBP13 - 2019
MacOS Big Sur 11.1
Python 3.8.2

Error when building wheel:
```
(virgo_venv) Otgers-MacBook-Pro:virgo-client otger$ pip install git+https://github.com/pybluez/pybluez.git
Collecting git+https://github.com/pybluez/pybluez.git
  Cloning https://github.com/pybluez/pybluez.git to /private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p
Requirement already satisfied: pyobjc-core>=6 in /Users/otger/development/virgo/virgo_venv/lib/python3.8/site-packages (from PyBluez==0.30) (7.0.1)
Requirement already satisfied: pyobjc-framework-Cocoa>=6 in /Users/otger/development/virgo/virgo_venv/lib/python3.8/site-packages (from PyBluez==0.30) (7.0.1)
Building wheels for collected packages: PyBluez
  Building wheel for PyBluez (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/otger/development/virgo/virgo_venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/setup.py'"'"'; __file__='"'"'/private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-wheel-wm7jx2d1
       cwd: /private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/
  Complete output (487 lines):
  Command line invocation:
      /Applications/Xcode.app/Contents/Developer/usr/bin/xcodebuild install -project macos/LightAquaBlue/LightAquaBlue.xcodeproj -scheme LightAquaBlue DSTROOT=/private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos INSTALL_PATH=/ DEPLOYMENT_LOCATION=YES
  
  Build settings from command line:
      DEPLOYMENT_LOCATION = YES
      DSTROOT = /private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos
      INSTALL_PATH = /
  
  note: Using new build system
  note: Building targets in parallel
  note: Planning build
  note: Constructing build description
  warning: Traditional headermap style is no longer supported; please migrate to using separate headermaps and set 'ALWAYS_SEARCH_USER_PATHS' to NO. (in target 'LightAquaBlue' from project 'LightAquaBlue')
  CreateBuildDirectory /private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos (in target 'LightAquaBlue' from project 'LightAquaBlue')
      cd /var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos/LightAquaBlue
      builtin-create-build-directory /private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos
  
  CreateBuildDirectory /Users/otger/Library/Developer/Xcode/DerivedData/LightAquaBlue-dpkqraonugmgfjgpmdlfcbbvdrzu/Build/Intermediates.noindex/ArchiveIntermediates/LightAquaBlue/IntermediateBuildFilesPath (in target 'LightAquaBlue' from project 'LightAquaBlue')
      cd /var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos/LightAquaBlue
      builtin-create-build-directory /Users/otger/Library/Developer/Xcode/DerivedData/LightAquaBlue-dpkqraonugmgfjgpmdlfcbbvdrzu/Build/Intermediates.noindex/ArchiveIntermediates/LightAquaBlue/IntermediateBuildFilesPath
  
  CreateBuildDirectory /Users/otger/Library/Developer/Xcode/DerivedData/LightAquaBlue-dpkqraonugmgfjgpmdlfcbbvdrzu/Build/Intermediates.noindex/ArchiveIntermediates/LightAquaBlue/BuildProductsPath (in target 'LightAquaBlue' from project 'LightAquaBlue')
      cd /var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos/LightAquaBlue
      builtin-create-build-directory /Users/otger/Library/Developer/Xcode/DerivedData/LightAquaBlue-dpkqraonugmgfjgpmdlfcbbvdrzu/Build/Intermediates.noindex/ArchiveIntermediates/LightAquaBlue/BuildProductsPath
  
  SymLink /private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos/LightAquaBlue.framework/Resources Versions/Current/Resources (in target 'LightAquaBlue' from project 'LightAquaBlue')
      cd /var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos/LightAquaBlue
      /bin/ln -sfh Versions/Current/Resources /private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos/LightAquaBlue.framework/Resources
  
  SymLink /private/var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos/LightAquaBlue.framework/LightAquaBlue Versions/Current/LightAquaBlue (in target 'LightAquaBlue' from project 'LightAquaBlue')
      cd /var/folders/9v/psr5rmy91691r58d86b3750c0000gn/T/pip-req-build-zed9y39p/macos/LightAquaBlue
[... cut ...]  -I/Users/otger/Library/Developer/Xcode/DerivedData/LightAquaBlue-dpkqraonugmgfjgpmdlfcbbvdrzu/Build/Intermediates.noindex/ArchiveIntermediates/LightAquaBlue/IntermediateBuildFilesPath/LightAquaBlue.build/Release/LightAquaBlue.build/DerivedSources-normal/arm64 -I/Users/otger/Library/Developer/Xcode/DerivedData/LightAquaBlue-dpkqraonugmgfjgpmdlfcbbvdrzu/Build/Intermediates.noindex/ArchiveIntermediates/LightAquaBlue/IntermediateBuildFilesPath/LightAquaBlue.build/Release/LightAquaBlue.build/Derived

rgov, 2020-12-21:
```
LightAquaBlue/BBServiceAdvertiser.m:157:9: error: implicit declaration of function 'IOBluetoothRemoveServiceWithRecordHandle' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
          return IOBluetoothRemoveServiceWithRecordHandle(handle);
                 ^
  1 error generated.
```

I looks like the `IOBluetoothRemoveServiceWithRecordHandle` function was removed. There is a comment above the call that says:

```
    // TODO: We should switch to using [IOBluetoothSDPServiceRecord removeServiceRecord]
    // but we don't know how to get an IOBluetoothSDPServiceRecord instance from a handle.
```

You could try just commenting out line 157 of that file, and replace it with `return kIOReturnUnsupported`.

mskoric, 2020-12-21:
hi,

had the same error and fixed it with

```
MacOsx catalina 10.15.7
python version 3.9.1

#pip uninstall pybluez
#git clone https://github.com/pybluez/pybluez.git
#cd pybluez;
# git diff macos/LightAquaBlue/BBServiceAdvertiser.m
diff --git a/macos/LightAquaBlue/BBServiceAdvertiser.m b/macos/LightAquaBlue/BBServiceAdvertiser.m
index f429914..5d07556 100644
--- a/macos/LightAquaBlue/BBServiceAdvertiser.m
+++ b/macos/LightAquaBlue/BBServiceAdvertiser.m
@@ -154,7 +154,8 @@ static NSDictionary *fileTransferProfileDict;
 {
     // TODO: We should switch to using [IOBluetoothSDPServiceRecord removeServiceRecord]
     // but we don't know how to get an IOBluetoothSDPServiceRecord instance from a handle.
-       return IOBluetoothRemoveServiceWithRecordHandle(handle);
+       // return IOBluetoothRemoveServiceWithRecordHandle(handle);
+       return kIOReturnUnsupported;
 }

pip install .

```

otger, 2020-12-22:
Thank you both! It worked (I do not use that functionality)

bpelloux, 2021-07-28:
hey everyone, i'm facing this issue as well and I tried to do what @mskoric said but I don't really get what you are doing after the cd pybluez command.. could you detail me what the git command do?

guilhermePaciulli, 2021-08-20:
@bpelloux he showed the diff he made so the project could compile

commit aada11c9 (2019-12-24): Build LightAquaBlue.framework when installed by recent versions of pip. Fixes #285