pybluez/pybluez #287: Allow sending mutable bytes-like objects

dogtopus, 2019-12-18:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** Linux (could be more, further investigation needed)
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.) Any
 - **Python Version:** 3.8
 - **PyBluez Version:** master (30060cd)

## Issue

Trying to send a memoryview object using `BluetoothSocket.send()` API results in an exception which states that only read-only bytes-like objects are expected.

A quick look at the `bluez/btmodule.c` reveals that the method expects `s#` type args as the buffer. According to the python documentation `s#` only accepts read-only (non-mutable) str or bytes-like object i.e. practically only str and bytes. One way to resolve this is to switch to `s*` or `y*` so that the API also accepts mutable bytes-like objects (like memoryview).

rgov, 2019-12-18:
Thanks for looking into the root cause of the issue. If you want to submit a pull request for this change, please test your solution and do so, and I will accept the change.

dogtopus, 2019-12-19:
Looks like the Windows binding (both widcomm and msbt) have the same problem. They all accepts `s#` as the parameter for write/send method.

I'll send a PR for BlueZ binding soon. Fix for Windows binding might take a while since I need to make a test environment first.

PR #288 (2020-01-14): Use buffer interface for send, sendall and sendto methods in BlueZ binding
Partially fixes #287.

Tested and seems to work fine.
closed 2020-01-14 by rgov