pybluez/pybluez #298: Latest pybluez 0.23 failes to look up deps when updating lockfile using pipenv

arves, 2020-01-02:
Dev system is Ubuntu 18.04 running virtualenv with python 3.6.9

Log: 
(qmachina) arves@NinjaBox:~/PycharmProjects/qmachina (develop) $ pipenv install pybluez
Installing pybluez…
✔ Installation Succeeded 
Pipfile.lock (69001b) out of date, updating to (a80a31)…
Locking [dev-packages] dependencies…
Locking [packages] dependencies…
✘ Locking Failed! 
Traceback (most recent call last):
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/resolver.py", line 126, in <module>
    main()
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/resolver.py", line 119, in main
    parsed.requirements_dir, parsed.packages)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/resolver.py", line 85, in _main
    requirements_dir=requirements_dir,
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/resolver.py", line 69, in resolve
    req_dir=requirements_dir
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/utils.py", line 726, in resolve_deps
    req_dir=req_dir,
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/utils.py", line 480, in actually_resolve_deps
    resolved_tree = resolver.resolve()
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/utils.py", line 385, in resolve
    results = self.resolver.resolve(max_rounds=environments.PIPENV_MAX_ROUNDS)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/resolver.py", line 102, in resolve
    has_changed, best_matches = self._resolve_one_round()
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/resolver.py", line 206, in _resolve_one_round
    for dep in self._iter_dependencies(best_match):
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/resolver.py", line 301, in _iter_dependencies
    dependencies = self.repository.get_dependencies(ireq)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/repositories/pypi.py", line 234, in get_dependencies
    legacy_results = self.get_legacy_dependencies(ireq)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/repositories/pypi.py", line 426, in get_legacy_dependencies
    results, ireq = self.resolve_reqs(download_dir, ireq, wheel_cache)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/repositories/pypi.py", line 297, in resolve_reqs
    results = resolver._resolve_one(reqset, ireq)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/resolve.py", line 260, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/resolve.py", line 213, in _get_abstract_dist_for
    self.require_hashes
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/operations/prepare.py", line 294, in prepare_linked_requirement
    abstract_dist.prep_for_dist(finder, self.build_isolation)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/operations/prepare.py", line 127, in prep_for_dist
    self.req.run_egg_info()
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/req/req_install.py", line 474, in run_egg_info
    command_desc='python setup.py egg_info')
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/utils/misc.py", line 705, in call_subprocess
    % (command_desc, proc.returncode, cwd))
pipenv.patched.notpip._internal.exceptions.InstallationError: Command "python setup.py egg_info" failed with error code 1 in /tmp/tmpeb1x_r57build/pyobjc-core/
File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/resolver.py", line 126, in <module>
    main()
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/resolver.py", line 119, in main
    parsed.requirements_dir, parsed.packages)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/resolver.py", line 85, in _main
    requirements_dir=requirements_dir,
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/resolver.py", line 69, in resolve
    req_dir=requirements_dir
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/utils.py", line 726, in resolve_deps
    req_dir=req_dir,
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/utils.py", line 480, in actually_resolve_deps
    resolved_tree = resolver.resolve()
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/utils.py", line 385, in resolve
    results = self.resolver.resolve(max_rounds=environments.PIPENV_MAX_ROUNDS)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/resolver.py", line 102, in resolve
    has_changed, best_matches = self._resolve_one_round()
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/resolver.py", line 206, in _resolve_one_round
    for dep in self._iter_dependencies(best_match):
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/resolver.py", line 301, in _iter_dependencies
    dependencies = self.repository.get_dependencies(ireq)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/repositories/pypi.py", line 234, in get_dependencies
    legacy_results = self.get_legacy_dependencies(ireq)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/repositories/pypi.py", line 426, in get_legacy_dependencies
    results, ireq = self.resolve_reqs(download_dir, ireq, wheel_cache)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/piptools/repositories/pypi.py", line 297, in resolve_reqs
    results = resolver._resolve_one(reqset, ireq)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/resolve.py", line 260, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/resolve.py", line 213, in _get_abstract_dist_for
    self.require_hashes
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/operations/prepare.py", line 294, in prepare_linked_requirement
    abstract_dist.prep_for_dist(finder, self.build_isolation)
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/operations/prepare.py", line 127, in prep_for_dist
    self.req.run_egg_info()
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/req/req_install.py", line 474, in run_egg_info
    command_desc='python setup.py egg_info')
  File "/home/arves/virtualenvs/qmachina/lib/python3.6/site-packages/pipenv/patched/notpip/_internal/utils/misc.py", line 705, in call_subprocess
    % (command_desc, proc.returncode, cwd))
pipenv.patched.notpip._internal.exceptions.InstallationError: Command "python setup.py egg_info" failed with error code 1 in /tmp/tmpeb1x_r57build/pyobjc-core/

Suspect that architecture is non critical, Darwin on ubuntu system for instance. Installation is ok, but failes to update lock file using pipenv as seen in the log. 

Using pipenv install pybluez="0.22" everything is still working as it should.

rgov, 2020-01-02:
It says that `python setup.py egg_info` is failing but it seemed to work for me on Ubuntu Bionic.

I do notice if I run it verbosely, it says that it requires pyobjc even on Ubuntu, which is not intentional.

rgov, 2020-01-02:
Installing with pip works fine, not sure what Pipenv is doing.

rgov, 2020-01-02:
Seems to be related to https://github.com/pypa/pipenv/issues/3187

closed 2020-02-08 by stale[bot]