pybluez/pybluez #300: "OSError: A request to send or receive data was disallowed" on Windows

bmeatayi, 2020-01-07:
## System

 - **Operating System:** Windows 10 Home
 - **Hardware:** Laptop with internal Bluetooth module
 - **Python Version:** 3.6.9 (Anaconda)
 - **PyBluez Version:** 0.22

## Issue

I used to work with pybluez on Windows 10 without any problem, but on my new laptop with Windows 10, I am getting this error:

```
Traceback (most recent call last):
  File "D:/Projects/examples/pybluez.py", line 35, in <module>
    sock.connect((host, port))
  File "C:\Users\mh_a\Anaconda3\envs\myenv\lib\site-packages\bluetooth\msbt.py", line 72, in connect
    bt.connect (self._sockfd, addr, port)
OSError: A request to send or receive data was disallowed because the socket had already been shut down in that direction with a previous shutdown call.
```

As it is not possible to debug _msbt.py functions in Windows, it is not clear to me what is the cause of this error. I tried turning off antivirus firewall (McAfee) but it didn't help.

Do you have any idea about the cause of this error?
Thanks in advance!

Here is the code I am running (rfcomm-client.py).
```
import sys

import bluetooth

addr = "00:13:43:6E:2F:A2"

print("Searching for SampleServer on {}...".format(addr))

# search for the SampleServer service
uuid = "1101"
service_matches = bluetooth.find_service(uuid=uuid, address=addr)

if len(service_matches) == 0:
    print("Couldn't find the SampleServer service.")
    sys.exit(0)

first_match = service_matches[0]
port = first_match["port"]
name = first_match["name"]
host = first_match["host"]

print("Connecting to \"{}\" on {}".format(name, host))

# Create the client socket
sock = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
sock.connect((host, port))

print("Connected. Type something...")
while True:
    data = input()
    if not data:
        break
    sock.send(data)

sock.close()
```

rgov, 2020-01-08:
Can you try the latest release, 0.23?

bmeatayi, 2020-01-08:
> Can you try the latest release, 0.23?

With the latest release, it doesn't even reach the socket connection step and gives an error in `find_service` method.
```
Traceback (most recent call last):
  File "C:/Users/mh_a/Desktop/Projects/github/examples/pybluez.py", line 44, in <module>
    service_matches = bluetooth.find_service(uuid=uuid, address=addr)
  File "C:\Users\mh_a\Anaconda3\envs\explorepy-win10\lib\site-packages\bluetooth\msbt.py", line 243, in find_service
    dresults = bt.find_service (addr, uuidstr)
OSError: A
```

rgov, 2020-01-08:
Perhaps this is what's happening, and the error message for Bluetooth is different than what we're displaying:

```
10058 WSAESHUTDOWN (“The L2CAP channel disconnected by remote peer”)
```

Are other Bluetooth functions of the new laptop working, besides PyBluez?

dogtopus, 2020-01-09:
>With the latest release, it doesn't even reach the socket connection step and gives an error in find_service method.

I guess #269 broke a lot more things than I previously expected. This could be another encoding issue.

bmeatayi, 2020-01-09:
> ing, and the error message for Bluetooth is different than what we're displaying:
> 
> ```
> 10058 WSAESHUTDOWN (“The L2CAP channel disconnected by remote peer”)
> ```
> 

Well, according to the [Windows Sockets Error Codes](https://docs.microsoft.com/en-us/windows/win32/winsock/windows-sockets-error-codes-2), the message for *10058 WSAESHUTDOWN* is 
```
Cannot send after socket shutdown.
A request to send or receive data was disallowed because the socket had already been shut down in that direction with a previous shutdown call. 
By calling shutdown a partial close of a socket is requested, which is a signal that sending or receiving, or both have been discontinued.
```
which is the message also shown the python. I wonder what the cause of the socket shot down is. 

> Are other Bluetooth functions of the new laptop working, besides PyBluez?
Yes, I haven't faced any other problems so far (e.g in file transfer).

rgov, 2020-01-14:
Does the patch in #286 make any difference to this example for you? It allegedly resolves some sort of Python 3 error.

bmeatayi, 2020-01-14:
> Does the patch in #286 make any difference to this example for you? It allegedly resolves some sort of Python 3 error.

I got this error while installing this patch.
```
(test-env) C:\Users\mh_at>pip install git+https://github.com/asaba96/pybluez.git@example-fixes
Collecting git+https://github.com/asaba96/pybluez.git@example-fixes
  Cloning https://github.com/asaba96/pybluez.git (to revision example-fixes) to c:\users\mh_at\appdata\local\temp\pip-req-build-mtv2i41x
  Running command git clone -q https://github.com/asaba96/pybluez.git 'C:\Users\mh_at\AppData\Local\Temp\pip-req-build-mtv2i41x'
  Running command git checkout -b example-fixes --track origin/example-fixes
  Branch 'example-fixes' set up to track remote branch 'example-fixes' from 'origin'.
  Switched to a new branch 'example-fixes'
Building wheels for collected packages: PyBluez
  Building wheel for PyBluez (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\Users\mh_at\Anaconda3\envs\test-env\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\mh_at\\AppData\\Local\\Temp\\pip-req-build-mtv2i41x\\setup.py'"'"'; __file__='"'"'C:\\Users\\mh_at\\AppData\\Local\\Temp\\pip-req-build-mtv2i41x\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\mh_at\AppData\Local\Temp\pip-wheel-ub3dqip5' --python-tag cp36
       cwd: C:\Users\mh_at\AppData\Local\Temp\pip-req-build-mtv2i41x\
  Complete output (72 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.6
  creating build\lib.win-amd64-3.6\bluetooth
  copying bluetooth\ble.py -> build\lib.win-amd64-3.6\bluetooth
  copying bluetooth\bluez.py -> build\lib.win-amd64-3.6\bluetooth
  copying bluetooth\btcommon.py -> build\lib.win-amd64-3.6\bluetooth
  copying bluetooth\macos.py -> build\lib.win-amd64-3.6\bluetooth
  copying bluetooth\msbt.py -> build\lib.win-amd64-3.6\bluetooth
  copying bluetooth\widcomm.py -> build\lib.win-amd64-3.6\bluetooth
  copying bluetooth\__init__.py -> build\lib.win-amd64-3.6\bluetooth
  Fixing build\lib.win-amd64-3.6\bluetooth\ble.py build\lib.win-amd64-3.6\bluetooth\bluez.py build\lib.win-amd64-3.6\bluetooth\btcommon.py build\lib.win-amd64-3.6\bluetooth\macos.py build\lib.win-amd64-3.6\bluetooth\msbt.py build\lib.win-amd64-3.6\bluetooth\widcomm.py build\lib.win-amd64-3.6\bluetooth\__init__.py
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  Fixing build\lib.win-amd64-3.6\bluetooth\ble.py build\lib.win-amd64-3.6\bluetooth\bluez.py build\lib.win-amd64-3.6\bluetooth\btcommon.py build\lib.win-amd64-3.6\bluetooth\macos.py build\lib.win-amd64-3.6\bluetooth\msbt.py build\lib.win-amd64-3.6\bluetooth\widcomm.py build\lib.win-amd64-3.6\bluetooth\__init__.py
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  running build_ext
  building 'bluetooth._msbt' extension
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "C:\Users\mh_at\AppData\Local\Temp\pip-req-build-mtv2i41x\setup.py", line 176, in <module>
      zip_safe=zip_safe)
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\site-packages\setuptools\__init__.py", line 145, in setup
      return distutils.core.setup(**attrs)
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\core.py", line 148, in setup
      dist.run_commands()
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\dist.py", line 955, in run_commands
      self.run_command(cmd)
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\dist.py", line 974, in run_command
      cmd_obj.run()
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\site-packages\wheel\bdist_wheel.py", line 192, in run
      self.run_command('build')
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\dist.py", line 974, in run_command
      cmd_obj.run()
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\command\build.py", line 135, in run
      self.run_command(cmd_name)
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\dist.py", line 974, in run_command
      cmd_obj.run()
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\site-packages\setuptools\command\build_ext.py", line 84, in run
      _build_ext.run(self)
    File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\command\build_ext.py", line 339, in run
      self.build_extensions()
[... 81 lines cut ...]
        _build_ext.run(self)
      File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\command\build_ext.py", line 339, in run
        self.build_extensions()
      File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\command\build_ext.py", line 448, in build_extensions
        self._build_extensions_serial()
      File "C:\Users\mh_at\Anaconda3\envs\test-env\lib\distutils\command\build_ext.py", line 473, in _build_extensions_serial
        self.build_extension(ext)
[... cut ...]
ERROR: Command errored out with exit status 1: 'C:\Users\mh_at\Anaconda3\envs\test-env\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\mh_at\\AppData\\Local\\Temp\\pip-req-build-mtv2i41x\\setup.py'"'"'; __file__='"'"'C:\\Users\\mh_at\\AppData\\Local\\Temp\\pip-req-build-mtv2i41x\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\mh_at\AppData\Local\Temp\pip-record-kts055aw\install-record.txt' --single-version-externally-managed --compile Check the logs for full command output.
```

bmeatayi, 2020-01-15:
> ## e
> I used to work with pybluez on Windows 10 without any problem, but on my new laptop with Windows 10, I am getting this error:
>
> ```
> Traceback (most recent call last):
>   File [...]

I tested with a few other laptops and it seems that the laptops with Realtek Bluetooth adapters have this issue. The script can be run without any problem on laptops with Intel Bluetooth adapter (tested on 5 different laptops - two of them with Realtek and three with Intel Bluetooth modules).

rgov, 2020-01-15:
Thats very interesting. PyBluez is just talking to the Windows BT APIs.

What about #224? Does that change (might have to apply it manually) affect anything?

bmeatayi, 2020-01-15:
> Thats very interesting. PyBluez is just talking to the Windows BT APIs.
> 
> What about #224? Does that change (might have to apply it manually) affect anything?

I installed with `Bthprops` but nothing changed. Unlike that issue, I have both `Irprops` and `Bthprops` in my Windows.

bmeatayi, 2020-01-21:
Trying to connect to the BT device with Boost library in C++ and MATLAB's instrument control toolbox also failed. I managed to connect to BT devices using a Bluetooth dongle. I am almost sure that the root of the problem is Realtek driver in Windows.

closed 2020-02-27 by stale[bot]