pybluez/pybluez #302: Segmentation fault via the methods `.get_l2cap_options()` and `.set_l2cap_mtu()`

b1ack0wl, 2020-01-08:
I'm working on a PR, but I wanted to at least file this bug to prevent duplicates from being submitted.

The following statements will cause a segmentation fault

```
import bluetooth

sock = bluetooth.BluetoothSocket(bluetooth.L2CAP)
sock.get_l2cap_options()
```
Attaching GDB shows that the crash is happening within `PyObject_Free` which is called via this statement here: https://github.com/pybluez/pybluez/blob/master/bluez/btmodule.c#L791.

rgov, 2020-01-08:
Thank you for working on a PR. I'll merge it when you've figured it out and it will make it into the next release.

b1ack0wl, 2020-01-08:
I fixed it, but could someone else double check my work?

I changed the following statement in `btmodule.c`
 * `PyObject *buf = PyUnicode_FromStringAndSize((char *)NULL, buflen);` to` PyObject *buf = PyBytes_FromStringAndSize((char *)NULL, buflen);` and the crash no longer happens. I also verified the syscall to `getsockopt` works fine via `strace`

b1ack0wl, 2020-01-08:
PR Opened: https://github.com/pybluez/pybluez/pull/303

closed 2020-01-08 by rgov