pybluez/pybluez #307: OSError: An invalid argument was supplied when using bluetooth.find_service

CTColeman65, 2020-01-21:
## System

 - **Operating System: Windows 10 Home
 - **Hardware: PC internal Bluetooth (asus zenbook ux430ua)
 - **Python Version: 3.7.4
 - **PyBluez Version: 0.23

## Issue

I am trying to execute the sample code but the `bluetooth.find_service` method is throwing the error 
`OSError: An invalid argument was supplied.`

The code I am running is simply
```
import sys
import bluetooth
uuid = "b378db85-4ec3-4daa-828e-1b99607bd6a0"
addr = None

connection = bluetooth.find_service(uuid=uuid, address=addr)
```

sd554, 2020-01-21:
I had this error, but I was able to fix it by switching to Python 3.5.4 and using the precompiled wheel file [here](https://www.lfd.uci.edu/~gohlke/pythonlibs/#pybluez) named `PyBluez‑0.22‑cp35‑none‑win_amd64.whl` to install PyBluez.

rgov, 2020-01-21:
This is failing inside https://github.com/pybluez/pybluez/blame/c435ae3ca48b2b10336dff4f467ce69475ba192c/msbt/_msbt.c#L580

There were changes recently:

- https://github.com/pybluez/pybluez/commit/0091913294350e49981a4981e40f568519fe63d0
- https://github.com/pybluez/pybluez/commit/2b200c2c7e8cb836c8a72339f647eca941a4d8a8

I would be curious if reverting either of those changes and building pybluez resolves the issue for you.

@dogtopus Any thoughts here?

gik98, 2020-02-05:
I can confirm the bug, both in 0.23 and the current master, with Python 3.8.1 on Windows 10 installed from the binary release on python.org.

gik98, 2020-02-05:
An update: following @rgov hint, I reverted both 0091913 and 2b200c2.

Now the function _kinda_ works, meaning that no exception is raised. Still, there seems to be some encoding issue preventing the name filter to function properly (I didn't try uuid and address).

If I execute:
```
matches = bluetooth.find_service(name="SPP_TEST")
```
no device is found. If I remove the name filter, than it finds the device, but it displays a wrong name: `b'SPP_TEST\x00'` instead of just `SPP_TEST`.

Now this might be an easy bug to solve for those that are familiar with the interworks between Windows C and Python, but I am not, so I'll just leave it here.

rgov, 2020-02-05:
Can you narrow down which of those two commits caused the regression?

gik98, 2020-02-05:
Yes, it's `2b200c2`. Apart from reverting, I copy-pasted the `port3.h` to the top of `_msbt.c`, beacuse the compiler complained about not being able to find that header file.

rgov, 2020-02-05:
Paging @hugovk. Could you help us understand this regression?

rgov, 2020-02-05:
@gik98 Actually, the change in 2b200c2 came after 0.23 was released to PyPI. So if you are seeing the issue with 0.23, simply reverting that commit shouldn't fix it, should it?

gik98, 2020-02-05:
It shouldn't and it doesn't, my bad. I started from a fresh clone and reproduced the following:
 * reverting `2b200c2` only has no effect
 * reverting `0091913` causes the following exception:
```
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "app\gui\gui.py", line 99, in fn_scan_devices
  File "app\device\bluetooth.py", line 19, in bt_find_devices
  File "site-packages\bluetooth\msbt.py", line 248, in find_service
  File "site-packages\bluetooth\btcommon.py", line 334, in sdp_parse_raw_record
  File "site-packages\bluetooth\btcommon.py", line 251, in sdp_parse_size_desc
TypeError: a bytes-like object is required, not 'str'
```
 * reverting both commits produces the behaviour described above. I froze that version [here](https://github.com/Kerr-srl/pybluez/tree/win10-sdp-regression)

hugovk, 2020-02-05:
`git bisect` could help out here: https://thoughtbot.com/blog/git-bisect

gik98, 2020-02-06:
An update: neither the BluetoothSocket connect function works, here the output
```
Exception in Tkinter callback
Traceback (most recent call last):
  File "tkinter\__init__.py", line 1705, in __call__
  File "app\gui\gui.py", line 117, in <lambda>
  File "app\gui\gui.py", line 138, in cb_btconnect
  File "app\device\bluetooth.py", line 61, in connect
  File "site-packages\bluetooth\msbt.py", line 98, in connect
OSError: 湁椠癮污摩愠杲浵湥⁴慷⁳畳灰楬摥മ
```
Where the OSError it is not chinese (or whatever those characters are), but just wrong encoding. Dropping that string into a UTF16 to UTF8 converter results in the string
" An invalid argument was supplied. "

Something is clearly wrong in everything dealing with string manipulation between Python 3 strings, Python 3 bytearrays and Windows C strings & byte [].

rgov, 2020-02-06:
Do you think you can figure out how to use a debugger break on this OSError being generated by the C extension and get a backtrace?

ricardoV94, 2020-09-09:
I am having the same issue on Windows 10, neither connect nor find_service are working. In addition I am getting the error messages truncated, observing only  `OSEroor: A`

gsakshi997, 2020-10-18:
Was anyone able to find a solution to this issue?
I am facing the same issue while I'm trying to make an RFCOMM connection.

dmata24, 2020-10-22:
Same here, I'm unable to make the connection between 2 laptops using PyBluez, when I run the client code I got the error: "Exception has occurred: OSError" on the "connect" line.
This is the code that I'm using for the client-server application 

Server code 
```
import Bluetooth 

server_sock  = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
port = 5
server_sock.bind(("", port))
server_sock.listen(1)

print("Waiting for connection")
    
client_sock, address = server_sock.accept()
print("Accepted connection from ", address)

data = client_sock.recv(1024)
print ("received [%s]" %data)

client_sock.close()
server_sock.close()
```

```
Client code
import Bluetooth 
sock = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
sock.connect(("B0:FC:36:16:D3:EE", 5))
sock.send("Hello")
sock.close
```

gsakshi997, 2020-10-23:
I was able to solve mine by using PyBluez-win10

jaafar-EL, 2021-01-19:
please i still have the same issue, i keep getting this error :
Traceback (most recent call last):
  File "C:\Users\User\PycharmProjects\pythonProject\main.py", line 12, in <module>
    services = bluetooth.find_service()
  File "C:\Users\User\PycharmProjects\pythonProject\venv\lib\site-packages\bluetooth\msbt.py", line 243, in find_service
    dresults = bt.find_service (addr, uuidstr)
OSError: Un argument non valide a été fourni.

if anyone can help me with this problem, I tried to install the latest version of pybluez library, but nothing changed.

wiseOsprey1, 2021-05-08:
Has this been resolved?  I'm seeing a similar issue when calling bluetooth.advertise_service():

Traceback (most recent call last):
  File "C:\Users\jonsc\OneDrive\Documents\Lawnmower\EV3PythonSocketServer.py", line 15, in <module>
    bluetooth.advertise_service( server_sock, "FooBar Service", uuid )
  File "C:\Users\jonsc\AppData\Local\Programs\Python\Python39\lib\site-packages\bluetooth\msbt.py", line 219, in advertise_service
    sock._sdp_handle = bt.set_service_raw (sock._raw_sdp_record, True)
OSError: An invalid argument was supplied.

Ripraskel, 2021-05-12:
Using `bluetooth.discover_devices` does work and gives me my mobile's Mac address

However, I get the error stated above when trying to use `bluetooth.find_service(None, None, addr)`
'addr' being my mobile Mac address I'd just got from `bluetooth.discover_devices`

dogtopus, 2021-06-17:
This particular issue should be resolved by the new PR. However I'm sure there are still similar unconverted `const? char *`/`LPC?STR` usages littered everywhere in the code. Need to do a deep cleanup sometime later.

PR #405 (2021-07-21): msbt: Correct more partial LPSTR to LPWSTR migrations.
Fixes #307.

~~This is not the end of it. We likely need to sweep through the whole _msbt.c codebase and move all LPSTR uses to LPWSTR in order to prevent issues like this from happening again.~~ I scanned through the code quickly and it seems that this is the last unconverted LPSTR usage. Let's hope that this is actually the last one.
commit d040baf7 (2021-07-21): msbt: Correct more partial LPSTR to LPWSTR migrations.

Fixes #307.

This is not the end of it. We likely need to sweep through the whole _msbt.c codebase and move all LPSTR uses to LPWSTR in order to prevent issues like this from happening again.
closed 2021-07-21 by rgov