pybluez/pybluez #308: Feature Request: Enable pin code verification when connecting to a device.

b1ack0wl, 2020-01-24:
**Note:** I'm just making this since I would like to take a stab at implementing this.

**Issue:** When trying to connect to a device that requires a pin code to be verified an exception is thrown saying `Permission Denied` (tested in Linux). This error is a little confusing since it seems like a user permission issue, but it's due to the `.connect()` method not allowing pin codes to be verified or used.

**Workaround:** Use `bluetoothctl` to pair to a device and verify the pincode via its CLI. Then, you can connect to any L2CAP PSM or RFCOMM port your heart desires.

**TODO:** Implement something within the `.connect()` method to handle connections that require a pin code to be verified. (e.g. car infotainment)

b1ack0wl, 2020-02-19:
I have not forgotten about this ticket. I was busy preparing for my talk at OffensiveCon and put this on the back burner. I will work on this and submit a PR within the next week.

rgov, 2020-02-19:
No worries. Which talk? I heard it was a good conference.

b1ack0wl, 2020-02-20:
My talk was titled "Don't forget to SUBSCRIBE" which talked about a heap-based overflow within Broadcom's upnp library specifically within the GENA stack. The exploit was fun to talk about since finding primitives like heap spraying and such took a lot of manual work. OffensiveCon is one of the only conferences that I'd recommend since everyone there is very passionate about finding vulns and deving out exploits. I honestly think it's the only conference I'll go to from now on since no one there had an ego and it was very nice to be there. I even made some new friends! :)

Anyways, I'll be working on adding the pincode verification feature within the `connect()` method since that's where it's currently dying for me. I also have a Windows VM and can use a USB bluetooth adapter via USB pass-through to test out the functionality. The only OS I cannot test on is OSX since I do not own any apple products.

closed 2020-03-28 by stale[bot]

rgov, 2020-03-28:
Looking forward to the pull request when you've got it working!