pybluez/pybluez #309: macOS discover_devices returns empty

mike-pt, 2020-01-30:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** macOS
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:** 3.7
 - **PyBluez Version:** 0.23

## Issue

```
>>> import bluetooth
>>> bluetooth.discover_devices(lookup_names=True)
[]
```

I have quite a few devices nearby (some connected to the system but not all)

rgov, 2020-01-31:
Please be more specific, what version of macOS are you using? What hardware is it? Are you using the built-in bluetooth chip or an external one?

Note that BTLE devices are not supported by PyBluez on macOS so it would have to be an older Bluetooth device.

mike-pt, 2020-01-31:
Hey sorry this is on a Macbook Pro (2017) running Catalina 10.15.2

here's some more info from systrem report with devices now connected (note that I'm trying to find others not listed here, none even support LE)

```
Apple Bluetooth Software Version:	7.0.2f4
  Hardware, Features and Settings:
  Name:	MBP
  Address:	00-00-00-00-00-00
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Broadcom
  Transport:	UART
  Chipset:	20703A2
  Firmware Version:	v91 c4301
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x05AC
  Product ID:	0x007E
  Bluetooth Core Spec:	4.2 (0x8)
  HCI Revision:	0x10CD
  LMP Version:	4.2 (0x8)
  LMP Subversion:	0x225B
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Portable
  Composite Class Of Device:	0x38010C
  Device Class (Major):	0x01
  Device Class (Minor):	0x03
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
  Devices (Paired, Configured, etc.):
  Keyboard K370/K375:
  Address:	00-00-00-00-00-00
  Random Address:	Yes
  Resolvable Address:	No
  Services:	
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Manufacturer:	Nordic Semiconductor (0x7, 0x5A)
  Bluetooth Core Spec:	4.1
  Vendor ID:	0x046D
  Product ID:	0xB34C
  Class of Device:	Low Energy
  AFH:	On
  AFH Map:	ffbfffff1f
  RSSI:	-54
  Role:	Central
  Connection Mode:	Active Mode
  Interval:	0 ms
  Name_of_device1:
  Address:	00-00-00-00-00-00
  Random Address:	Yes
  Resolvable Address:	No
  Services:	
  Paired:	Yes
  Configured:	Yes
  Connected:	Yes
  Bluetooth Core Spec:	4.0
  Vendor ID:	0x248A
  Product ID:	0x8266
  Class of Device:	Low Energy
  AFH:	On
  AFH Map:	ffbfffff1f
  RSSI:	-61
  Role:	Central
  Connection Mode:	Active Mode
  Interval:	0 ms
 Name_of_device2:
  Address:	00-00-00-00-00-00
  Major Type:	Miscellaneous
  Minor Type:	Unknown
  Services:	
  Paired:	No
  Configured:	Yes
  Connected:	No
  Bluetooth Core Spec:	1.0b
  Class of Device:	0x00 0x00 0x0000
  Name_of_device3:
  Address:	00-00-00-00-00-00
  Major Type:	Audio
  Minor Type:	Headset
  Services:	MVIEW, Advanced Audio, Advanced Audio, Hands-Free unit
  Paired:	Yes
  Configured:	Yes
  Connected:	No
  Bluetooth Core Spec:	4.2
  Class of Device:	0x04 0x01 0x2C0404
  EDR Supported:	Yes
  eSCO Supported:	Yes
  SSP Supported:	Yes
```

mike-pt, 2020-01-31:
hum Actually looks like today I can get it to list my smartphone, there was a Catalina update today but they didn't list anything remotely related to bluetooth.

I was also seeing issues connecting the Wireless phones (via the system settings) which is the last device on that list and rebooted, maybe it was just that.

In any case since I get at least one working I'm gonna close this for now and reopen if it makes sense.

btw is there a way to list already paired devices ? couldn't find much info on that

closed 2020-01-31 by mike-pt