pybluez/pybluez #312: Can gattlib be rolled forward to a newer version?

askpatrickw, 2020-02-06:
## System

 - **Operating System:** Linux ubuntu 5.3.0-1017-raspi2 #19-Ubuntu SMP Thu Jan 16 18:25:50 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
 - **Hardware:** Raspberry Pi 3b+
 - **Python Version:** Python 3.7.5
 - **PyBluez Version:** PyBluez 0.30

## Issue
I was having a heck of time getting pybluz[ble] installed due to issue with the gattlib install.
I was able to build gatlib from source at version 0.20200122 and then install pybluez from source as well when I overrided the gattlib version.

Is there some known reason gattlib is locked down to 0.20150805

Setup.py change I used once I had gattlib built and installed:
```
-      extras_require={'ble': ['gattlib==0.20150805']},
+      extras_require={'ble': ['gattlib==0.20200122']},
```

rgov, 2020-02-06:
Good question. If you give it more testing, I'd be happy to upgrade it (5 years old, yikes). I don't use Linux so I can't test it.

We also have this fork of https://github.com/pybluez/pygattlib which I'm skeptical of. Is it still needed? Is it even used?

askpatrickw, 2020-02-07:
> We also have this fork of https://github.com/pybluez/pygattlib which I'm skeptical of. Is it still needed? Is it even used?

I hasn't been updated in about 2 years... AND you're not specifying your version be installed from source so I'd say its not being used.

##Script Testing

Checkmark means I have run it, the comments reflect the actual pass\fail or anecdotal comments about what I observed. All were run with `sudo python3 <script>.py`

### advanced

- [x] inquiry-with-rssi.py - Failed #317 but not related to gattlib version
- [ ] l2-mtu.py 
- [ ] l2-unreliable-client.py 
- [ ] l2-unreliable-server.py  
- [x] read-local-bdaddr.py  - Pass
- [x] write-inquiry-scan.py - Pass

### ble

While I mark these as Pass, there appears to be some instability. For example read_name will give errors `RuntimeError: Could not update HCI connection: Input/output error` and then work fine the next ten times. I'm not sure if that is due to lack of error handling in the example and the general intermittency of Bluetooth or another issue exists.

- [x] beacon.py  - Pass
- [x] beacon_scan.py  - Pass
- [x] read_name.py  - Pass
- [x] scan.py  - Pass

### bluezchat

- [ ] bluezchat.py

### simple

- [x] asynchronous-inquiry.py   - Failed #317  but not related to gattlib version
- [x] inquiry.py  - Pass
- [ ] l2capclient.py  
- [ ] l2capserver.py  
- [ ] rfcomm-client.py  
- [ ] rfcomm-server.py  
- [x] sdp-browse.py - Pass

rgov, 2020-02-07:
This was the incentive for the gattlib fork: https://github.com/pybluez/pybluez/pull/194

> Due to the issue mentioned in #173 and the fact that gattlib doesn't seem to be maintained anymore, I have forked the BitBucket pygattlib repo to danielunderwood/pygattlib to fix the boost dependencies and whatever else I come across. 

It looks like https://bitbucket.org/OscarAcena/pygattlib has seen some progress lately, but it is only available as a Debian package, and not from PyPI.

rgov, 2020-02-07:
I marked our fork as abandoned. I think this should be rolled forward. Please see if you encounter any other issues with the remaining examples and send a pull request.

askpatrickw, 2020-02-16:
I have moved on to a different library. From my notes above someone else should be able to complete testing. Best of luck!

rgov, 2020-02-16:
Ok, thanks. Which library did you find fit your needs better?

askpatrickw, 2020-02-16:
Bleson, it worked right out the gate. The parsing needs some help, but I could see BLE data on first attempt on Ubuntu on Raspberry Pi 3b+.