pybluez/pybluez #317: Unicode Errors when running examples

askpatrickw, 2020-02-06:
## System

 - **Operating System:** uname -a
`Linux ubuntu 5.3.0-1017-raspi2 #19-Ubuntu SMP Thu Jan 16 18:25:50 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux`
 - **Hardware:** Raspberry Pi 3b+
 - **Python Version:** Python 3.7.5
 - **PyBluez Version:** PyBluez 0.30
- **Gattlib Version:** gattlib 0.20200122 <-- Built from source
- **apt list bluez:** `bluez/eoan-updates,now 5.50-0ubuntu5 arm64 [installed]`

## Issue

Might be related to #67 

examples/advanced/inquiry-with-rssi.py
```
Current inquiry mode is 1
Traceback (most recent call last):
  File "inquiry-with-rssi.py", line 159, in <module>
    device_inquiry_with_with_rssi(sock)
  File "inquiry-with-rssi.py", line 84, in device_inquiry_with_with_rssi
    bluez.hci_filter_all_events(flt)
UnicodeDecodeError:
   'utf-8' codec can't decode byte 0xff in position 4: invalid start byte
```

examples/simple/asynchronous-inquiry.py
```
Traceback (most recent call last):
  File "asynchronous-inquiry.py", line 61, in <module>
    d.find_devices(lookup_names=True)
  File "/usr/local/lib/python3.7/dist-packages/bluetooth/bluez.py", line 434, in find_devices
    _bt.hci_filter_all_events (flt)
UnicodeDecodeError:
  'utf-8' codec can't decode byte 0xff in position 4: invalid start byte
```

rgov, 2020-02-06:
Sounds like `hci_filter_all_events` should be using `bytes` instead of `unicode`.

askpatrickw, 2020-02-07:
I'm not a C developer, so hopefully someone else watching the repo sees this and can help out.

rgov, 2020-02-07:
Is it enough to just switch the gattlib version in setup.py and then build the package, and then run the example? Do I need any other dependencies?

closed 2020-03-15 by stale[bot]