pybluez/pybluez #318: "OSError: A" in bt.connect() on Windows

JamoDevNich, 2020-02-09:
## System

 - **Operating System:** Windows 7
 - **Hardware:** Intel Wireless AC 7260 with Bluetooth 4.0
 - **Python Version:** 3.6.5
 - **PyBluez Version:** 0.23

## Issue
When attempting to connect to a device (LYWSD03MMC) using the code below, an OSError occurs:
```Python
import bluetooth;
from time import sleep;
print("creating socket");
socket_ = bluetooth.BluetoothSocket(bluetooth.RFCOMM);
print("connecting");
socket_.connect(("A4:C1:38:DF:xx:xx", 1));
print("waiting on connected...");
while not socket_.connected():
    pass;
print("connected successfully");
sleep(4);
print("now will disconnect");
socket_.close();
while socket_.connected():
    pass;
print("successfully disconnected");
```
Output:
```
creating socket
connecting
Traceback (most recent call last):
  File "lywsblehelper.py", line 33, in <module>
    socket_.connect(("A4:C1:38:DF:xx:xx", 1));
  File "C:\Users\yardefaragunle\AppData\Local\Programs\Python\Python36\lib\site-packages\bluetooth\msbt.py", line 96, in connect
    bt.connect (self._sockfd, addr, port)
OSError: A
```

However, the device enquiry sample code (pybluez/examples/simple/inquiry.py) works as expected, showing both Classic BT and BLE devices:

Output:
```
performing inquiry...
found 9 devices
  30:xx:xx:xx:xx:xx - Lamp-007

  41:xx:xx:xx:xx:xx - iPhone
  54:xx:xx:xx:xx:xx - iPhone
  58:xx:xx:xx:xx:xx -
  84:xx:xx:xx:xx:xx - Nicholas Elliott (Galaxy S5)
  A4:C1:38:DF:xx:xx - LYWSD03MMC
  B8:xx:xx:xx:xx:xx -
  DF:xx:xx:xx:xx:xx - MX Master
  F8:xx:xx:xx:xx:xx -
```

Some observations that might help??
- My computer is able to connect to BLE devices, such as a mouse
- Another computer running Debian with a similar Intel adapter can connect to and communicate with the BLE device (LYWSD03MMC)

I was hoping to be able to develop a cross-platform client for this BLE device

rgov, 2020-02-09:
Can you try installing from this Git repository? I don't think it fixes your issue but it will fix the display of the error message.

If PyBluez doesn't work (Windows support is broken right now, I think), you can also check out PyQt5's Bluetooth support, which is cross platform.

JamoDevNich, 2020-02-10:
Fails looking for winsock2.h - I think that's a part of a windows dev kit, I'll try again after locating it and installing on my machine...

```
Collecting git+https://github.com/pybluez/pybluez
  Cloning https://github.com/pybluez/pybluez to c:\users\yardef~1\appdata\local\temp\pip-req-build-y162bt0i
Installing collected packages: PyBluez
  Running setup.py install for PyBluez: started
    Running setup.py install for PyBluez: finished with status 'error'
    Complete output from command c:\users\yardefaragunle\appdata\local\programs\python\python36\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\YARDEF~1\\AppData\\Local\\Temp\\pip-req-build-y162bt0i\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\YARDEF~1\AppData\Local\Temp\pip-record-t2o0jrbw\install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build\lib.win-amd64-3.6
    creating build\lib.win-amd64-3.6\bluetooth
    copying bluetooth\ble.py -> build\lib.win-amd64-3.6\bluetooth
    copying bluetooth\bluez.py -> build\lib.win-amd64-3.6\bluetooth
    copying bluetooth\btcommon.py -> build\lib.win-amd64-3.6\bluetooth
    copying bluetooth\macos.py -> build\lib.win-amd64-3.6\bluetooth
    copying bluetooth\msbt.py -> build\lib.win-amd64-3.6\bluetooth
    copying bluetooth\widcomm.py -> build\lib.win-amd64-3.6\bluetooth
    copying bluetooth\__init__.py -> build\lib.win-amd64-3.6\bluetooth
    Fixing build\lib.win-amd64-3.6\bluetooth\ble.py build\lib.win-amd64-3.6\bluetooth\bluez.py build\lib.win-amd64-3.6\bluetooth\btcommon.py build\lib.win-amd64-3.6\bluetooth\macos.py build\lib.win-amd64-3.6\bluetooth\msbt.py build\lib.win-amd64-3.6\bluetooth\widcomm.py build\lib.win-amd64-3.6\bluetooth\__init__.py
    Skipping optional fixer: buffer
    Skipping optional fixer: idioms
    Skipping optional fixer: set_literal
    Skipping optional fixer: ws_comma
    Fixing build\lib.win-amd64-3.6\bluetooth\ble.py build\lib.win-amd64-3.6\bluetooth\bluez.py build\lib.win-amd64-3.6\bluetooth\btcommon.py build\lib.win-amd64-3.6\bluetooth\macos.py build\lib.win-amd64-3.6\bluetooth\msbt.py build\lib.win-amd64-3.6\bluetooth\widcomm.py build\lib.win-amd64-3.6\bluetooth\__init__.py
    Skipping optional fixer: buffer
    Skipping optional fixer: idioms
    Skipping optional fixer: set_literal
    Skipping optional fixer: ws_comma
    running build_ext
    building 'bluetooth._msbt' extension
    creating build\temp.win-amd64-3.6
    creating build\temp.win-amd64-3.6\Release
    creating build\temp.win-amd64-3.6\Release\msbt
    C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.14.26428\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ic:\users\yardefaragunle\appdata\local\programs\python\python36\include -Ic:\users\yardefaragunle\appdata\local\programs\python\python36\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.14.26428\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" /Tcmsbt\_msbt.c /Fobuild\temp.win-amd64-3.6\Release\msbt\_msbt.obj
    _msbt.c
    msbt\_msbt.c(5): fatal error C1083: Cannot open include file: 'winsock2.h': No such file or directory
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Enterprise\\VC\\Tools\\MSVC\\14.14.26428\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
    
    ----------------------------------------
```

A VM running win10 1809 LTSB with various dev tools installed gives a different error - "fatal error LNK1158: cannot run 'rc.exe'":
[image]

JamoDevNich, 2020-02-10:
0.30 successfully installed on my machine from this repo - slightly different error now.

The BLE device's 'Connected' icon shows up for a second and a half before disappearing.

```
creating socket
connecting
Traceback (most recent call last):
  File "C:\Users\yardefaragunle\Documents\GitHub\LYWS-BLE-Library\lywsblehelper.py", line 32, in <module>
    socket_.connect(("A4:C1:38:DF:xx:xx", 1));
  File "C:\Users\yardefaragunle\AppData\Local\Programs\Python\Python36\lib\site-packages\bluetooth\msbt.py", line 96, in connect
    bt.connect (self._sockfd, addr, port)
OSError: A socket operation was attempted to an unreachable network.
```

closed 2020-02-10 by JamoDevNich