pybluez/pybluez #319: BLE - RuntimeError: Set scan parameters failed (are you root?)

alborzs, 2020-02-14:
I am running the BLE example code on a Raspberry Pi 4 with no luck.
It is complaining regardless if I run as sudo or not, see below:

```
$ python3 testBLE.py
Traceback (most recent call last):
  File "testBLE.py", line 5, in <module>
    devices = service.discover(2)
RuntimeError: Set scan parameters failed (are you root?)
```

```
$ sudo python3 testBLE.py
Traceback (most recent call last):
  File "testBLE.py", line 5, in <module>
    devices = service.discover(2)
RuntimeError: Set scan parameters failed (are you root?)
```
The device is working fine, so that is not the problem.

```
$ sudo hciconfig lescan
hci0:   Type: Primary  Bus: UART
        BD Address: DC:A6:32:2A:F6:A9  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:2517 acl:0 sco:0 events:105 errors:0
        TX bytes:2564 acl:0 sco:0 commands:58 errors:0
```

rgov, 2020-02-14:
Hi @alborzs, the reason we have the issue template text that you deleted is because it helps understand what you are running on. At a minimum:

* Is this Raspbian Buster?
* What version of PyBluez are you using?

If you are not using 0.23, please try that version. If you are, please try installing the latest code from this repository.

alborzs, 2020-02-15:
Sorry about that, the reason why I omitted that part is that I though it was a given that I am on the latest release of Rasbian Buster and PyBluez.
No reason to be running old codes and submitting issues.

closed 2020-03-23 by stale[bot]

xkonni, 2020-03-26:
happening on my xps 13 7390 (i7-10510u), Intel bt device 8087:0029 as well.
- archlinux 
- linux 5.5.11
- bluez 5.54-1
- bluez-libs 5.54-1
- bluez-utils 5.54-1

any idea how to debug this? I can get a "somehow similar" message when trying to disable le:
```bash
$ sudo btmgmt le off                                                  
Set Low Energy for hci0 failed with status 0x0a (Busy)
```

So mabye some process is actively blocking le. Not sure which one though. All else seems to be fine.

autostock, 2022-10-28:
This worked for me. From
https://unix.stackexchange.com/questions/96106/bluetooth-le-scan-as-non-root

```
sudo apt-get install libcap2-bin # was already installed
sudo setcap 'cap_net_raw,cap_net_admin+eip' `which python3` # problem: python3 was a symlink to python3.7
ls -l /usr/bin/python3 # where is the symlink going
sudo setcap 'cap_net_raw,cap_net_admin+eip' `which python3.7`
python3 scanBLE.py

```