pybluez/pybluez #320: Python 2.7 - v0.23 already unsupported by Py2.7

TAbdiukov, 2020-02-26:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

**Python Version:** 2.7
**PyBluez Version:** 0.23

## Issue

I know Py2.7 is dead, and rightfully so, but here you go. The alleged last version of pybluez is [https://github.com/pybluez/pybluez/issues/266](0.23). But from the discussion there it is (and probably was, at the time) unclear whether the last Py2.7 compatible version was intended to be 0.22 or 0.23 . As it currectly stands, if I do,
```
pip2 install pybluez
```
It installs v0.23 . After installation, I try to import another package (info removed to avoid unintentional spam) requiring pybluez and here is that happens
```
>> import (anotherpackage)
  File "C:\Python27\lib\site-packages\another\common.py", line 33, in <module>
    from bluetooth import BluetoothSocket, RFCOMM
  File "C:\Python27\lib\site-packages\bluetooth\__init__.py", line 282, in <modu
le>
    """
AttributeError: attribute '__doc__' of 'instancemethod' objects is not writable
```

This, along with non-clarity in the [discussion about retiring Py2.7](https://github.com/pybluez/pybluez/issues/266) seems to imply **v0.22** was intended to be the last Py2.7 version all along. And indeed, happen I install v0.22 , the problem disappears

Given all this, do you believe we should remove v0.23 from allegedly being Py2.7-compatible on PyPI? Cheers

rgov, 2020-02-26:
These are the pains of a team of people who don't know how to release a complex Python package trying to do so...

I'd be happy to get 0.23 working on Python 2.7 (since it is also broken on Windows I think). Do you know how to fix it?

For what it's worth, we have a continuous integration system that did not catch this.

rgov, 2020-02-26:
Line that fails: https://github.com/pybluez/pybluez/blob/0998c40ebad69b3c8e2b3e90f1fd0c0879ce6a57/bluetooth/__init__.py#L274

kopop8, 2020-03-11:
Thanks i searched for hours how to get pybluez working. This should really be documented better.
I deleted pybluez and installed v 0.22 by doing `pip install 'pybluez==0.22'` for anyone wondering.

TAbdiukov, 2020-03-14:
@rgov Unfortunately have zero idea. I needed pyBluez to work to get another related, but overly tightly coupled-to-pyBluez, package to work to have some of its unrelated to BT functionality I needed.

@kopop8 It's all good, but I myself not entirely sure (and as I noted, perhaps the developers themselves also) whether v0.23 simply contains an overlooked bug, or v0.22 was intended to be the last for Py2.7. Regardless, as you noted the workaround is to install pyBluez via `pip install pybluez==0.22`

rgov, 2020-05-14:
The easiest solution right now is to install 0.22 or to comment out the offending documentation, since they're not important, and do a custom installation.

@beadysea helped with the documentation here but I guess we missed it. Is there an alternative we could do?

mgb158, 2021-02-22:
Pragmatic fix, in __init__.py test for python version and only set up the documentation for >=3:

```
if sys.version >= '3':
    BluetoothSocket.dup.__doc__ =\
        """Duplicate the socket

        Returns
        -------
        BluetoothSocket
            A new :class:`BluetoothSocket` connected to the same system resource.
            
        """
    BluetoothSocket.accept.__doc__ = \
        """Accept a connection.

        Returns
        -------
        tuple
            A tuple containing a :class:`BluetoothSocket` and a Bluetooth address.
```
etc