pybluez/pybluez #323: BLE: RuntimeError: Set scan parameters failed (are you root?)

xkonni, 2020-03-26:
XPS13 7390 fails to scan for BLE devices

error:
```
RuntimeError: Set scan parameters failed (are you root?)
```

## System

- archlinux 
- linux 5.5.11
- bluez 5.54-1
- bluez-libs 5.54-1
- bluez-utils 5.54-1

## Issue

This is happening on my xps 13 7390 (i7-10510u), with hci0 (internal Intel 8087:0029)
it does work with another device hci1 (usb Asus 0b05:17cb)

```bash
$ sudo python3 
Python 3.8.2 (default, Feb 26 2020, 22:21:03) 
[GCC 9.2.1 20200130] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from bluetooth.ble import DiscoveryService
>>> ds0 = DiscoveryService("hci0")
>>> ds0.discover(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Set scan parameters failed (are you root?)
>>> ds1 = DiscoveryService("hci1")
>>> ds1.discover(1)
{'xxxxxx': '', ...}
```

Not sure what that actually means. Maybe something with the intel driver?
If you need more information/logs, I'm happy to provide them.

isfahany, 2020-04-22:
As I know, this thing happen when you try to access hardware(I/O) thing which is essentialy need to accessed by root. you run python3 as sudoers, but We didint know about your line script. I'm researching it rightnow, brb

xkonni, 2020-04-23:
well yes, in theory. but i do run it as root and it does work with module b (usb Asus 0b05:17cb) and not with module a (internal Intel 8087:0029).

closed 2020-05-30 by stale[bot]