pybluez/pybluez #326: discover_devices returns incomplete list of devices

demongolem, 2020-04-05:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** 
Windows 10
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
Atmotube Pro Device
 - **Python Version:**
3.6
 - **PyBluez Version:**
0.23

## Issue

Please provide information necessary to reproduce the issue, including sample
code.

If you have long logs, please post them on https://gist.github.com and link to
the Gist here.

The list of devices returned by discover_devices is incomplete.  I cannot fully determine which devices pybluez does and does not return so easily because the name is None for all 3 devices returned.  I do not the address of the atmotube device, however it is not on the list of 3 returned decvices.  So I call discover_devices and iterate over all the returned address values printing out each address.

cozy-chestnut, 2020-04-25:
Add lookup_names attribute when you make the function call, you should be able to get a list with address and name.
```
nearby_devices = bluetooth.discover_devices(lookup_names = True)
```

closed 2020-06-02 by stale[bot]

hatkidchan, 2020-11-17:
~~Sorry, but how you managed to run Windows 10 on Raspberry Pi?~~

Also, I had the same issue on Arch Linux with python 3.8.6 (I'm using integrated wifi+bluetooth card (`0a5c:21b4` Broadcom Corp. BCM2070 Bluetooth 2.1 + EDR)), BUT `bluetoothctl` can show devices, when PyBlueZ can not. I have no idea, have that issue something simillar with mine, but who knows ¯\_(ツ)_/¯