pybluez/pybluez #327: Fail to recv data from client (Windows 10)

sgallou, 2020-04-08:
## System

 - **Operating System:** Windows10
 - **Hardware:** PC with external Bluetooth adaptor (Fujitsu USB Stick v2.0)
 - **Python Version:** 3.8.0
 - **PyBluez Version:** 0.23+ (head of master branch)

## Issue

I made a simple RFCOMM client, but fail to receive from (note that send works well). Here is the code :

```
socket = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
socket.connect(("D8:A0:1D:56:3E:52", 1))
while True:
    data = socket.recv(6)
    for d in data:
        print(hex(ord(d)))
```

Other side send 7 bytes.
Behaviour is different when I put 6 or 10 as recv argument.

# recv(6)

Result is :
```
0x0
0x0
0x0
0x0
0x6ce8
0x6e0b
Fatal Python error: Immortal interned string died.
Python runtime state: initialized
```
...with callstack of all threads ([full log](https://gist.github.com/sgallou/41d9ecbc4d30420da664de88709455fd))

# recv(10)

Got the 'bad argument to internal function' error at data receiving.
([full log](https://gist.github.com/sgallou/d8249e6cacddc828a7ed6690b016c1fb))

rgov, 2020-04-08:
Seems like there might be heap corruption / use-after-free going on? Is it possible to build with Address Sanitizer or Valgrind or something?

sgallou, 2020-04-15:
Hello, thanks for your answer. I installed MSVS 2019, but I don't know how to build pybluez. I clone it (master branch), but I don't find any makefile or similar.
If I'm able to build it, I will debug inside and see exactly what is failing.
So is there any documentation about building pybluez ?

rgov, 2020-04-15:
Usually you just run "python setup.py bdist_wheel" if you've got Visual
Studio and the Windows SDK installed already.

PR #329 (2020-04-23): Return Bytes instead of Unicode
Fix the msbt_recv function to return bytes instead of unicode.  This commit combined with the fix in PR #226 got my usb bluetooth dongle to receive bytes on a Windows 10 machine.  This might fix #327.
closed 2020-04-23 by rgov