pybluez/pybluez #331: Different return types from discover_devices() (macOS & Windows)

cozy-chestnut, 2020-04-25:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System: macOS and Windows
 - **Hardware: Desktop and Laptop
 - **Python Version: Both Python 3.7.7
 - **PyBluez Version: Both 0.23

## Issue

Code:
```
import bluetooth

nearby_devices = bluetooth.discover_devices(lookup_names = True)

for addr, name in nearby_devices:
    print(type(addr), type(name))
```
On Windows, I got both <class 'str'>.
On macOS, I got <class 'bytes'> <class 'objc.pyobjc_unicode'>

rgov, 2020-04-25:
Good catch. Can you check the latest code from Git and see if it resolves the issue?

PyBluez has a different "backend" to talk to each OS's Bluetooth stack which is the main problem why this project is kind of a mess. I would not be surprised if there are many many cases where this is true.

I'd love to have a testing infrastructure that could catch this. But we can't test with real bluetooth devices in CI so it's hard to catch bugs.

cozy-chestnut, 2020-04-26:
@rgov Thanks for the follow-up. I have tried getting the latest version of PyBluez(0.30) in this main repo. It turns out it doesn't work on masOS. I got an ImportError for the LightAquaBlue framework.

Does the latest code require the LightAquaBlue framework to work?

cozy-chestnut, 2020-04-26:
@rgov By the way, I'm wondering if there is a complete PyBluez documentation exist somewhere? I'm trying to write some code to track the live signal strength to a targeted device. Something like a Bluetooth based proximity tracker. But couldn't find any resource to figure this out. Any help or recommendation will be appreciated.

rgov, 2020-04-26:
There's a current pull request that's open that's supposed to make LightAquaBlue work on macOS. You can also see how the continuous integration script builds for Mac.

Docs are supposed to be at https://pybluez.readthedocs.io.

cozy-chestnut, 2020-04-26:
Are you talking about the pull request posted by @tigerking? Isn't it haven't been reviewed yet?

rgov, 2020-04-26:
It hasn't been approved yet, no, but you can clone their repo (from https://github.com/tigerking/pybluez) and report if it works. That will help me decide whether to accept the pull request.

For what it's worth, I didn't write PyBluez and don't really know much about it, I just spend a small amount of my time accepting patches from people who are able to fix bugs on their system. I can't offer much support. I know slightly more about macOS than other systems.

cozy-chestnut, 2020-04-26:
Just tested it. Unfortunately, it doesn't fix the issue of load the lightblue framework. I have taken a look into that python site-packages directory. The LightAquaBlue.framewok is no longer there.

Went to LightBlue website and check out their installation for macOS. Looks like they no longer support the project. One of the macOS sdk their setup.py is referring to is too old. Can't even install it on macOS Catalina.

rgov, 2020-04-26:
How are you installing it? You don't need to go look for other repos, this one should have whatever you need.  Use `python setup.py bdist_wheel`.

cozy-chestnut, 2020-04-26:
I installed it from `pip3 install pybluez` for version 0.23
And got the latest one using `pip3 install --upgrade git+https://github.com/pybluez/pybluez.git` for version 0.30. This is the version giving me `cannot not load LightBlue framework` error.

cozy-chestnut, 2020-04-26:
@rgov Successfully got version 0.30 running on macOS Catalina, here is what I did:
1. Clone this repo to local
2. cd to the local copy
3. Run `python setup.py bidet_wheel`
4. Run `pip install ./` to install the downloaded repo.

Confirmed Version 0.30 is working fine now. However, the return type from discover_devices() are still different from what I got from Windows.

closed 2020-06-02 by stale[bot]