pybluez/pybluez #333: How can I read the all the broadcast data from the advertisement device (Manufacturer, Service...)?

Milan991, 2020-04-30:
## System

 - **Operating System: Raspbian Buster**
 - **Hardware: Raspberry Pi**
 - **Python Version: 3.6**

Is there a possibility to read whole broadcast packet from the advertisement device? I would like to read Manufacturer ID and other data that my beacon is sending inside the advertisement packet.

Thank you in advance,
Milan.

rgov, 2020-04-30:
You should first figure out whether the Linux Bluez API allows you to do this. http://people.csail.mit.edu/albert/bluez-intro/ may have some (potentially outdated) information.

If you can figure out the Linux API, we can see if Pybluez supports it yet. If it doesn't, maybe it's not so hard to add.

ukBaz, 2020-04-30:
The Linux DBUS API has some hooks in there:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/device-api.txt#n235
This will only show you one reading for the beacon so if the manufacturer data gets updated you will not see it.

There is a promising sounding method in the Adapter API to allow duplicates:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/adapter-api.txt#n107

However, there is some filtering goes on if the same device is seem multiple times to stop the dbus getting flooded. The switch to allow duplicates does not appear to be enabled in the kernel from what I've read on the mailing list:
https://marc.info/?l=linux-bluetooth&m=158257065411112&w=2

The following API has been added in the last couple of weeks, but I've not had time to look into it and if it solves the problem:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertisement-monitor-api.txt

Most of the solutions I've seen for reading beacons end up using very low-level HCI calls into the kernel. An example of such a library is:
https://github.com/frawau/aioblescan

Milan991, 2020-05-11:
Closing this issue as I managed to get the data by using https://github.com/singaCapital/BLE-Beacon-Scanner

closed 2020-05-11 by Milan991