pybluez/pybluez #334: Reading RSSI from specific device?

AbelValko, 2020-05-01:
## System

 - **Operating System:** Raspbian Buster
 - **Hardware:** Raspberry Pi
 - **Python Version:** 3.7

Is it possible to get an RSSI readout for a specific device address without having to scan for several seconds? Either after a connection is established or without. I am trying to continuously update the RSSI value as part of a larger script and so scanning for 10 seconds with for example device_inquiry_with_with_rssi is not an option. Thanks in advance!

laksh225, 2020-05-23:
Normally, protocols used for bluetooth connections are either RFCOMM or L2CAP. But, the protocol used to get the RSSI value from devices is HCI. So, you will have to make a seperate socket and then get the value. And I don't think there is any way to get it faster.

laksh225, 2020-05-24:
Look's like I was wrong.  [Bluetooth-proximity](https://github.com/ewenchou/bluetooth-proximity) does give RSSI value of specified device/s. And it's fast.

closed 2020-07-03 by stale[bot]