pybluez/pybluez #335: Add wrappers for whitelist functions

alessandro-allen, 2020-05-01:
Currently, there is no support for hci_le_add_white_list, hci_le_read_white_list_size, and hci_le_clear_white_list. btmodule.c should be extended so as to wrap these functions.

rgov, 2020-05-01:
Sounds like you know what needs to be changed. If you create a pull request I can merge it. But there are no developers implementing features.

alessandro-allen, 2020-05-01:
GitHub is telling me that I don't have write access to open a PR. Can you correct this for me?

rgov, 2020-05-01:
Did you create a fork? You fork the project, make your changes in your own fork (for which you have write access) and then you can initiate a pull request from your fork. 

We get pull requests from time to time and I’ve never had to change anyone’s permissions. 

Let me know if I can help further.

closed 2020-05-08 by alessandro-allen