pybluez/pybluez #338: Server example leaking channels on OSX

zellerman, 2020-05-09:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** Macos 10.15.3
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:** 3.7.4
 - **PyBluez Version:** 0.30 tigerking patch: d485fd6

## Issue

I'm running the sample server which listens on rfcomm, and I noticed that terminating the program results in leaking bluetooth channels. I.e. in _Waiting for connection on RFCOMM channel x_ ,  `x` gradually increments by one. But when I try to connect with a client (Android) to the server, the client logs that it connected to the earliest `x` channel.
I guess the program should release the socket even on sigint/sigterm/sigabrt etc.

zellerman, 2020-05-09:
Of course you could do something like this:
```
def sigterm_handler(sig, frame):
    server_sock.close()
    sys.exit(0)
```

But shouldn't this be a framework responsibility? Probably on system framework level.

rgov, 2020-05-09:
This would have to be handled by [LightAquaBlue](https://github.com/pybluez/pybluez/tree/master/macos) which is used on macOS. I don't know specifically it would have to be handled.

zellerman, 2020-05-10:
I see. (Anyway closing the socket on sigterm also seems to be ineffective in some cases. )
I'll probably try to dig deeper and see what's happening on the firmware level. Though I can't promise anything.

closed 2020-06-16 by stale[bot]