pybluez/pybluez #339: Error while installing pybluez-win10 downloaded and installed microsoft bulid tools

muzamilhuss, 2020-05-14:
Using cached PyBluez-win10-0.22.tar.gz (97 kB)
Building wheels for collected packages: pybluez-win10
  Building wheel for pybluez-win10 (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\users\mmh\pycharmprojects\untitled\venv\scripts\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\MMH\\AppData\\Lo
cal\\Temp\\pip-install-4rjv92ay\\pybluez-win10\\setup.py'"'"'; __file__='"'"'C:\\Users\\MMH\\AppData\\Local\\Temp\\pip-install-4rjv92ay\\pybluez-win10\\setup.py'"'
"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))'
 bdist_wheel -d 'C:\Users\MMH\AppData\Local\Temp\pip-wheel-mkey34uh'
       cwd: C:\Users\MMH\AppData\Local\Temp\pip-install-4rjv92ay\pybluez-win10\
  Complete output (48 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.7
  creating build\lib.win-amd64-3.7\bluetooth
  copying bluetooth\ble.py -> build\lib.win-amd64-3.7\bluetooth
  copying bluetooth\bluez.py -> build\lib.win-amd64-3.7\bluetooth
  copying bluetooth\btcommon.py -> build\lib.win-amd64-3.7\bluetooth
  copying bluetooth\macos.py -> build\lib.win-amd64-3.7\bluetooth
  copying bluetooth\msbt.py -> build\lib.win-amd64-3.7\bluetooth
  copying bluetooth\widcomm.py -> build\lib.win-amd64-3.7\bluetooth
  copying bluetooth\__init__.py -> build\lib.win-amd64-3.7\bluetooth
  Fixing build\lib.win-amd64-3.7\bluetooth\ble.py build\lib.win-amd64-3.7\bluetooth\bluez.py build\lib.win-amd64-3.7\bluetooth\btcommon.py build\lib.win-amd64-3.7\
bluetooth\macos.py build\lib.win-amd64-3.7\bluetooth\msbt.py build\lib.win-amd64-3.7\bluetooth\widcomm.py build\lib.win-amd64-3.7\bluetooth\__init__.py
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  Fixing build\lib.win-amd64-3.7\bluetooth\ble.py build\lib.win-amd64-3.7\bluetooth\bluez.py build\lib.win-amd64-3.7\bluetooth\btcommon.py build\lib.win-amd64-3.7\
bluetooth\macos.py build\lib.win-amd64-3.7\bluetooth\msbt.py build\lib.win-amd64-3.7\bluetooth\widcomm.py build\lib.win-amd64-3.7\bluetooth\__init__.py
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  running build_ext
  building 'bluetooth._msbt' extension
  creating build\temp.win-amd64-3.7
  creating build\temp.win-amd64-3.7\Release
  creating build\temp.win-amd64-3.7\Release\msbt
  C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.25.28610\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD "-IC:\Program
 Files (x86)\Microsoft SDKs\Windows\v10.0A\Include" -I.\port3 -Ic:\users\mmh\pycharmprojects\untitled\venv\include -IC:\Users\MMH\AppData\Local\Programs\Python\Pyt
hon37\include -IC:\Users\MMH\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.25.2
8610\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Fi
les (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10
\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /Tcmsbt\_msbt.c /Fobuild\temp.win-amd64-3.7\Release\msbt\_msb
t.obj
  _msbt.c
  msbt\_msbt.c(141): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
  msbt\_msbt.c(170): warning C4996: 'WSAStringToAddressA': Use WSAStringToAddressW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API wa
rnings
  msbt\_msbt.c(218): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
  msbt\_msbt.c(241): warning C4996: 'WSAStringToAddressA': Use WSAStringToAddressW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API wa
rnings
  msbt\_msbt.c(398): warning C4244: '=': conversion from 'SOCKET' to 'int', possible loss of data
  msbt\_msbt.c(391): warning C4996: 'WSADuplicateSocketA': Use WSADuplicateSocketW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API wa
rnings
  msbt\_msbt.c(395): warning C4996: 'WSASocketA': Use WSASocketW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings
  msbt\_msbt.c(411): warning C4244: 'initializing': conversion from 'SOCKET' to 'int', possible loss of data
  msbt\_msbt.c(553): warning C4996: 'WSAStringToAddressA': Use WSAStringToAddressW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API wa
rnings
  msbt\_msbt.c(631): warning C4244: 'initializing': conversion from 'SOCKET' to 'int', possible loss of data
  msbt\_msbt.c(660): warning C4996: 'WSALookupServiceBeginA': Use WSALookupServiceBeginW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated
API warnings
  msbt\_msbt.c(682): warning C4996: 'WSALookupServiceNextA': Use WSALookupServiceNextW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated AP
I warnings
  msbt\_msbt.c(803): warning C4311: 'type cast': pointer truncation from 'HANDLE' to 'unsigned long'
  msbt\_msbt.c(795): warning C4996: 'WSASetServiceA': Use WSASetServiceW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings
  msbt\_msbt.c(852): warning C4996: 'WSASetServiceA': Use WSASetServiceW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings
  C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.25.28610\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EM
[... 73 lines cut ...]
warnings
    msbt\_msbt.c(631): warning C4244: 'initializing': conversion from 'SOCKET' to 'int', possible loss of data
    msbt\_msbt.c(660): warning C4996: 'WSALookupServiceBeginA': Use WSALookupServiceBeginW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecate
d API warnings
    msbt\_msbt.c(682): warning C4996: 'WSALookupServiceNextA': Use WSALookupServiceNextW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated
API warnings
    msbt\_msbt.c(803): warning C4311: 'type cast': pointer truncation from 'HANDLE' to 'unsigned long'
    msbt\_msbt.c(795): warning C4996: 'WSASetServiceA': Use WSASetServiceW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings
    msbt\_msbt.c(852): warning C4996: 'WSASetServiceA': Use WSASetServiceW() instead or define _WINSOCK_DEPRECATED_NO_WARNINGS to disable deprecated API warnings
    C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.25.28610\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:
EMBED,ID=2 /MANIFESTUAC:NO "/LIBPATH:C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\Lib\x64" /LIBPATH:c:\users\mmh\pycharmprojects\untitled\venv\libs /LIBPAT
H:C:\Users\MMH\AppData\Local\Programs\Python\Python37\libs /LIBPATH:C:\Users\MMH\AppData\Local\Programs\Python\Python37 /LIBPATH:c:\users\mmh\pycharmprojects\untit
led\venv\PCbuild\amd64 "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.25.28610\lib\x64" "/LIBPATH:C:\Program Files (x86)
\Windows Kits\NETFXSDK\4.8\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10
\lib\10.0.19041.0\um\x64" WS2_32.lib Irprops.lib /EXPORT:PyInit__msbt build\temp.win-amd64-3.7\Release\msbt\_msbt.obj /OUT:build\lib.win-amd64-3.7\bluetooth\_msbt.
cp37-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.7\Release\msbt\_msbt.cp37-win_amd64.lib
    LINK : fatal error LNK1181: cannot open input file 'Irprops.lib'
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.25.28610\\bin\\HostX86\\x64\\link.exe' failed with exit
 status 1181
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'c:\users\mmh\pycharmprojects\untitled\venv\scripts\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0]
 = '"'"'C:\\Users\\MMH\\AppData\\Local\\Temp\\pip-install-4rjv92ay\\pybluez-win10\\setup.py'"'"'; __file__='"'"'C:\\Users\\MMH\\AppData\\Local\\Temp\\pip-install-4
rjv92ay\\pybluez-win10\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compil
e(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\MMH\AppData\Local\Temp\pip-record-rk1klyny\install-record.txt' --single-version-externally-managed -
-compile --install-headers 'c:\users\mmh\pycharmprojects\untitled\venv\include\site\python3.7\pybluez-win10' Check the logs for full command output.

rgov, 2020-05-14:
Please don't just paste the console barf when you get an error. There was an issue template that you deleted, which asked for important information.  It is also impossible to read because you did not apply proper formatting.

rgov, 2020-05-14:
Aside from that `pybluez-win10` is not an official library, try to just install `pybluez`.

muzamilhuss, 2020-05-14:
sorry for uploading the full console error detail but when i install the pybluez it give me an error when a try to connect a device
-------------------------------------'
Traceback (most recent call last):
  File "C:/Users/MMH/Desktop/new22.py", line 18, in <module>
    client_socket.connect((bd_addr,3))
  File "C:\Users\MMH\PycharmProjects\untitled\venv\lib\site-packages\bluetooth\msbt.py", line 96, in connect
    bt.connect (self._sockfd, addr, port)
OSError: A

charlamenan, 2020-06-01:
Hi muzamilhuss,

Did you get this working ,i am looking for same solution .
I would want to automate scanning and pairing of bluetooth headset in windows 10 .

charlamenan, 2020-06-03:
@muzamilhuss Did you get this working ?

muzamilhuss, 2020-06-03:
@chalamenan i found another way to connect with a Bluetooth device its very simple.. you can you serial module instead of pybluez..

muzamilhuss, 2020-06-03:
> @charlamenan  Did you get this working ?
 i found another way to connect with a Bluetooth device its very simple.. you can use serial module instead of pybluez..

charlamenan, 2020-06-03:
> > @charlamenan  Did you get this working ?
> > i found another way to connect with a Bluetooth device its very simple.. you can use serial module instead of pybluez..

@muzamilhuss ,Thanks for the reply .
I want to scan bluetooth devices and pair a device in windows10 .
The bluetooth device is a bluetooth headset or bluetooth speaker ,i would like only up to connection ,no need of sending message and receiving message .Can you please send example code since you have it already working for scanning and pairing part .Meanwhile i will go through the serial module .

charlamenan, 2020-06-03:
@muzamilhuss ,can i contact you through email.My email is charlamenan@gmail.com ,please provide yours .

muzamilhuss, 2020-06-03:
> @charlamenan i have sent you a mail

charlamenan, 2020-06-11:
Hi ,
Yes ,i could install with python3.7.4 and windows 10 .
with python3.8 version i get error during installation

francescomaldonato, 2020-06-11:
Hi,
Thank you for your reply. I have installed python 3.7.4 and I think I
installed all the dependencies but there is still something wrong.
I attached a screenshot of all the programs I installed for pybluez just to
show you and maybe you can notice something is missing.
Best regards

closed 2020-07-18 by stale[bot]

prusswan, 2024-06-22:
For "OSError: A" error on Python 3.7, I went back to 0.22 (do not use 0.23). Someone recommended building from latest master, that works too. So the conclusion is that 0.23 release is bad for Windows/Python 3.7

prusswan, 2024-06-22:
For "OSError: A" error on Python 3.7, I went back to 0.22 (do not use 0.23 - see #349). Someone recommended building from latest master, that works too. So the conclusion is that 0.23 release is bad for Windows/Python 3.7