pybluez/pybluez #342: Update readme -- Linux PyBluez requires libbluetooth-dev

Vbbab, 2020-05-18:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** Ubuntu 16.04 LTS
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.) PC
 - **Python Version:** 3.5.2
 - **PyBluez Version:** 0.23 (Tried to install)

## Issue
Compilation fails, complaining about a missing `bluetooth/bluetooth.h` header file which is not installed by default on Ubuntu 16.04 LTS

Please provide information necessary to reproduce the issue, including sample
code.
Simply running `pip install pybluez` (or pip3 for python3)

Partial output of `pip install pybluez`:
```
In file included from bluez/btmodule.c:20:0:
    bluez/btmodule.h:5:33: fatal error: bluetooth/bluetooth.h: No such file or directory
```

Fix:
1. Run `sudo apt-get install libbluetooth-dev`
2. Rerun `pip install pybluez`

rgov, 2020-05-18:
Thank you. Please submit a pull request clarifying the necessary packages in this file:

https://github.com/pybluez/pybluez/blob/master/docs/install.rst

laksh225, 2020-05-22:
Please close this issue as it is resolved.

Vbbab, 2020-05-22:
The dependency requirements have been updated.
The `install.rst` file was already updated, closing.

closed 2020-05-22 by Vbbab