pybluez/pybluez #344: Example script gives error

laksh225, 2020-05-23:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:*Ubuntu 20.04*
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:*3.8.2*
 - **PyBluez Version:*0.23*

## Issue

The example code for rfcomm-server gives error.. 

Traceback (most recent call last):
  File "/home/laksh/.local/lib/python3.8/site-packages/bluetooth/bluez.py", line 271, in advertise_service
    _bt.sdp_advertise_service (sock._sock, name, service_id, \
_bluetooth.error: (2, 'No such file or directory')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "rfcomm-server.py", line 20, in <module>
    bluetooth.advertise_service(server_sock, "SampleServer", service_id=uuid,
  File "/home/laksh/.local/lib/python3.8/site-packages/bluetooth/bluez.py", line 275, in advertise_service
    raise BluetoothError (*e.args)
bluetooth.btcommon.BluetoothError: [Errno 2] No such file or directory

vik0t0r, 2020-06-29:
possible duplicate with https://github.com/pybluez/pybluez/issues/161 ?

laksh225, 2020-06-30:
Ohh. Yes it is.. Thanks.

closed 2020-06-30 by laksh225

adityasharma-tech, 2025-06-24:
After starting `bluetoothd` with `compat` flag. Everything is fine.
```bash
sudo bluetoothd --nodetach --debug --compat
```