pybluez/pybluez #348: pybluez support for python3.8.X version

charlamenan, 2020-06-01:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** windows10
 - **Python Version:** python 3.8.3
 - **PyBluez Version:**PyBluez-0.23

## Issue
pip install pybluez gives the following error 
Collecting pybluez
  Using cached PyBluez-0.23.tar.gz (97 kB)
Using legacy setup.py install for pybluez, since package 'wheel' is not installed.
Installing collected packages: pybluez
    Running setup.py install for pybluez ... error
    ERROR: Command errored out with exit status 1:

charlamenan, 2020-06-01:
pybluez installation works with python3.7.4 version ,but i wanted it on python3.8.X versions .
could somebody help me please ?

rgov, 2020-06-01:
Try `pip install wheel` and then try again.

charlamenan, 2020-06-02:
After installing ,different issue is seen.can i get a .whl file for python3.8.X version .

WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x00000213813AA160>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed'))': /simple/pybluez/
**ERROR: Could not find a version that satisfies the requirement pybluez (from versions: none)
**ERROR: No matching distribution found for pybluez****

charlamenan, 2020-06-03:
@rgov ,Wheel is given for download that supports 3.7 version  https://pypi.org/project/PyBluez/#files
Can i get wheel for download that supports python 3.8 version

closed 2020-07-10 by stale[bot]

ricardoV94, 2020-09-09:
Reopening to mirror request from @charlamenan

sanjayv2002, 2023-03-08:
pybluez seems to be simple when it comes to implementation. Is there any alternatives for interfacing bluetooth devices using python?

rgov, 2023-03-08:
Qt for Python (formerly? alternatively? known as PyQt) does: https://doc.qt.io/qtforpython/examples/example_bluetooth__heartrate_game.html

dogtopus, 2023-03-08:
>pybluez seems to be simple when it comes to implementation

Actually there are quite a lot of things that makes it unnecessarily complicated to maintain in the long term (Bluetooth stack being highly OS-dependent without a clear common-ground interface being one)

sanjayv2002, 2023-03-09:
> Qt for Python (formerly? alternatively? known as PyQt) does: https://doc.qt.io/qtforpython/examples/example_bluetooth__heartrate_game.html

This seems like the thing I was looking for. I was looking for a way to read data from a Bluetooth device using python. (cuz I need to do lots of processing on the said data). Feel free to put suggestions guys for this task, Thanks in advance 

And I will try both of your suggestions as well. @dogtopus  and @rgov