pybluez/pybluez #349: Anyone got Bluetooth connection working in windows

charlamenan, 2020-06-10:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** windows 10
 - **Hardware:** Bluetooth speaker
 - **Python Version:**3.7
 - **PyBluez Version:**pybluez 0.23

## Issue
  File "C:\Program Files\Python37\lib\site-packages\bluetooth\msbt.py", line 96, in connect
    bt.connect (self._sockfd, addr, port)
OSError: A

My code :

import bluetooth
nearby_devices = bluetooth.discover_devices(lookup_names=True)
target_name = "Carvaan mini"
port = 1
for addr,name in nearby_devices:
    print(" {} - {}".format(name,addr))
    if target_name in name:
        print("target device found")
        target_addr=addr

# Create the client socket
sock = bluetooth.BluetoothSocket(bluetooth.RFCOMM)
sock.connect((target_addr, port))

dmeignan, 2020-07-09:
Got a similar issue.
You may try with pyBluez 0.22.

rayipoiod, 2020-07-28:
i finally got 0.22 and 0.22 (Pybluez-win10) working on win10.

what i did is: 
make a copy of bthprops.lib from where it is and rename it to IRPROPS.LIB (make sure you find the right one for me it is X64), then everything works like champ. 

what i believe the reason is:
since the IRPROPS.LIB is depreciated in the latest win10 sdk, so most of the pybluez version is not working out of the box. i have tried with the 0.3(latest commit), it can be installed, but not worked for my environment. 

#https://blogs.windows.com/windowsdeveloper/2019/11/19/windows-10-sdk-preview-build-19023-available-now/

closed 2020-09-03 by stale[bot]

hiaselhans, 2020-10-31:
with #226 merged in, is it possible to have a new release? 0.24?

mshaheerm, 2020-11-06:
@rayipoiod  where is this file and how to get pybluez 0.22 ? i am having the same issue but unable to resolve

fbouchet, 2020-11-26:
> 
> 
> @rayipoiod where is this file and how to get pybluez 0.22 ? i am having the same issue but unable to resolve

Try `pip install PyBluez==0.22 --user` - it solved the issue for me

prusswan, 2024-06-22:
Both 0.22 and latest master (pip install https://github.com/pybluez/pybluez.git#egg=pybluez) worked for me. 

Related: #318, #279