pybluez/pybluez #357: revc sometimes gets stuck

zlj-zz, 2020-08-20:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** linux
 - **Hardware:**  Raspberry Pi
 - **Python Version:** 3.7
 - **PyBluez Version:** 0.23

## Issue

```python
while file_len != 0:
    try:
        data = client_sock.recv(buffer)
   except bluetooth.BluetoothErrora as e:
        print(e)
        break
        f.write(data)
       file_len -= len(data)
```

I accept a file on the Raspberry Pi, about 6 M in size. In the process of receiving `recv` is easy to get stuck, I am sure there is data continue to be sent. Can you  help me?

ukBaz, 2020-08-20:
Are you trying to transfer a file?  This would be better done with the [Object Push Profile](https://www.bluetooth.com/specifications/profiles-overview/)?

There is some more information on how to do this at:
https://raspberrypi.stackexchange.com/a/114187/121848

zlj-zz, 2020-08-21:
thank you，i'll try it.

closed 2020-09-27 by stale[bot]