pybluez/pybluez #358: Set scan parameters failed (are you root?)

haonv206, 2020-09-15:
## System

 - **Operating System:** Raspbian / release: 10
 - **Hardware:** Raspberry Pi
 - **Python Version:** 3.5.9
 - **PyBluez Version:** 0.23

## Issue
I follow the sample code to scan for BLE device, but it gives the error: Set scan parameters failed (are you root?)
- **user**: pi
- **groups**: pi adm dialout cdrom sudo audio video plugdev games users input netdev bluetooth gpio i2c spi

```python
# bluetooth low energy scan
from bluetooth.ble import DiscoveryService
service = DiscoveryService()
devices = service.discover(2)
---------------------------------
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Set scan parameters failed (are you root?)
```

I searched around and found some information about rootless setup (https://www.home-assistant.io/integrations/bluetooth_le_tracker#rootless-setup), but the error is still the same

My capabilities are like these:
```bash
pi@raspberrypi:~/src $ sudo getcap `which python`
/home/pi/.pyenv/shims/python = cap_net_admin,cap_net_raw+eip
pi@raspberrypi:~/src $ sudo getcap `which python3`
/home/pi/.pyenv/shims/python3 = cap_net_admin,cap_net_raw+eip
pi@raspberrypi:~/src $ sudo getcap `which hcitool`
/usr/bin/hcitool = cap_net_raw+ep
```

closed 2020-09-16 by haonv206

vladyslav2, 2020-12-01:
@haonv206 if you fixed this, can you collaborate how?

haonv206, 2020-12-03:
@vladyslav2 I still can't run on 'pi' user, but 'root' user is ok