pybluez/pybluez #359: python3.7 win10  platform con't install pybluez-win10,even using \Microsoft Visual C++ Build Tools, error: command 'cl.exe' failed: No such file or directory

zcj2010lzh, 2020-09-20:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:**
 - **Hardware:** (e.g., Raspberry Pi, external Bluetooth adaptor, etc.)
 - **Python Version:**
 - **PyBluez Version:**

## Issue

Please provide information necessary to reproduce the issue, including sample
code.

If you have long logs, please post them on https://gist.github.com and link to
the Gist here.

rgov, 2020-09-20:
@zcj2010lzh Please install the latest `pybluez` package instead of `pybluez-win10`.

@KivenChen Could you please remove your unofficial package from PyPI? It was helpful for users when you posted it, but now it is a few years out of date. And it is generating a lot of confusion where people come to this project for help, not understanding that we did not make the package. We are happy to take your patch for Windows 10 if there is anything else we are missing.

closed 2020-10-28 by stale[bot]

KivenChen, 2020-10-28:
Sorry for all inconvenience caused. This "win10" package has been removed
because it is no longer active or useful.

There was once a path problem causing the package installation to fail when
Python is set up for a particular user (not for all). However, it should be
already fixed right now.