pybluez/pybluez #361: OSX 10.15.5, PyBlueZ, Python 3.7.3

scgrant327, 2020-09-27:
Exception has occurred: BluetoothError
[Errno -536870174] Error starting device inquiry
  File "/Users/owner/BTST/Web/ws/server/wx/stn/wsStation.py", line 123, in <module>
    nearby_devices = bluetooth.discover_devices(duration=8, lookup_names=True, flush_cache=True, lookup_class=False)

Why? What's wrong?

scgrant327, 2020-09-27:
Code:
```
import bluetooth

nearby_devices = bluetooth.discover_devices(duration=8, lookup_names=True, flush_cache=True, lookup_class=False)
print("Found {} devices.".format(len(nearby_devices)))

for addr, name in nearby_devices:
    print("  {} - {}".format(addr, name))
```

boaz-warshawsky, 2020-10-19:
I am also experiencing the same issue, any thoughts here?

networkextension, 2020-10-22:
install python-lightblue
$pip3 install python-lightblue

boaz-warshawsky, 2020-10-22:
Thanks @networkextension, **that** problem is resolved.

closed 2020-11-28 by stale[bot]

payasparab, 2023-02-02:
> Thanks @networkextension, **that** problem is resolved.
This worked for me

https://stackoverflow.com/a/74639096