pybluez/pybluez #365: pip3 install pybluez[ble] fails on Ubuntu20.04

paaguti, 2020-11-02:
*PyBluez is not under active development but we are seeking new contributors
to investigate bugs and submit patches.*

## System

 - **Operating System:** Ubuntu 20.04 desktop
 - **Hardware:** HP elitebook 840
 - **Python Version:** 3.8
 - **PyBluez Version:** 

## Issue
My question: is the [ble] option still required?

Apparently, the gattlib required is ancient and can't be used on modern Ubuntu's:
```
vibe-pro3 ➤ sudo -H python3 -m pip install pybluez\[ble\]                      
Requirement already satisfied: pybluez[ble] in /usr/local/lib/python3.8/dist-packages (0.23)
Collecting gattlib==0.20150805; extra == "ble"
  Using cached gattlib-0.20150805.tar.gz (1.7 MB)
Building wheels for collected packages: gattlib
  Building wheel for gattlib (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-47ejtasp/gattlib/setup.py'"'"'; __file__='"'"'/tmp/pip-install-47ejtasp/gattlib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-ex93lyyh
       cwd: /tmp/pip-install-47ejtasp/gattlib/
  Complete output (30 lines):
  running bdist_wheel
  running build
  running build_ext
  building 'gattlib' extension
  creating build
  creating build/temp.linux-x86_64-3.8
  creating build/temp.linux-x86_64-3.8/src
  creating build/temp.linux-x86_64-3.8/src/bluez
  creating build/temp.linux-x86_64-3.8/src/bluez/lib
  creating build/temp.linux-x86_64-3.8/src/bluez/attrib
  creating build/temp.linux-x86_64-3.8/src/bluez/src
  creating build/temp.linux-x86_64-3.8/src/bluez/src/shared
  creating build/temp.linux-x86_64-3.8/src/bluez/btio
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/gattservices.cpp -o build/temp.linux-x86_64-3.8/src/gattservices.o
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/beacon.cpp -o build/temp.linux-x86_64-3.8/src/beacon.o
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/bindings.cpp -o build/temp.linux-x86_64-3.8/src/bindings.o
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/gattlib.cpp -o build/temp.linux-x86_64-3.8/src/gattlib.o
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/bluez/lib/uuid.c -o build/temp.linux-x86_64-3.8/src/bluez/lib/uuid.o
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/bluez/attrib/gatt.c -o build/temp.linux-x86_64-3.8/src/bluez/attrib/gatt.o
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/bluez/attrib/gattrib.c -o build/temp.linux-x86_64-3.8/src/bluez/attrib/gattrib.o
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/bluez/attrib/utils.c -o build/temp.linux-x86_64-3.8/src/bluez/attrib/utils.o
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/bluez/attrib/att.c -o build/temp.linux-x86_64-3.8/src/bluez/attrib/att.o
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/bluez/src/shared/crypto.c -o build/temp.linux-x86_64-3.8/src/bluez/src/shared/crypto.o
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/bluez/src/log.c -o build/temp.linux-x86_64-3.8/src/bluez/src/log.o
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python3.8 -c src/bluez/btio/btio.c -o build/temp.linux-x86_64-3.8/src/bluez/btio/btio.o
  creating build/lib.linux-x86_64-3.8
  x86_64-linux-gnu-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.8/src/gattservices.o build/temp.linux-x86_64-3.8/src/beacon.o build/temp.linux-x86_64-3.8/src/bindings.o build/temp.linux-x86_64-3.8/src/gattlib.o build/temp.linux-x86_64-3.8/src/bluez/lib/uuid.o build/temp.linux-x86_64-3.8/src/bluez/attrib/gatt.o build/temp.linux-x86_64-3.8/src/bluez/attrib/gattrib.o build/temp.linux-x86_64-3.8/src/bluez/attrib/utils.o build/temp.linux-x86_64-3.8/src/bluez/attrib/att.o build/temp.linux-x86_64-3.8/src/bluez/src/shared/crypto.o build/temp.linux-x86_64-3.8/src/bluez/src/log.o build/temp.linux-x86_64-3.8/src/bluez/btio/btio.o -lglib-2.0 -lboost_python-py34 -lboost_thread -lbluetooth -o build/lib.linux-x86_64-3.8/gattlib.cpython-38-x86_64-linux-gnu.so
[... cut ...]
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-47ejtasp/gattlib/setup.py'"'"'; __file__='"'"'/tmp/pip-install-47ejtasp/gattlib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-90bxhveh/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.8/gattlib Check the logs for full command output.
vibe-pro3 ➤                                                                    

```

vladyslav2, 2020-12-01:
@paaguti you are missing `boost_python-py34` library try to look and install for boost dev or python-dev  package first

vladyslav2, 2020-12-01:
whole sequance:
```
sudo apt-get install libbluetooth-dev
sudo apt-get install libboost-python-dev
sudo apt-get install libboost-all-dev
git clone https://github.com/pybluez/pybluez
cd pybluez
pip3 install .
```

closed 2021-01-09 by stale[bot]